realsenseai/librealsense #4671: Problem/Issue : intelRealSense D435 Depth Frame not comming

TanmayPathak, 2019-08-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Operating System & Version |   Ubuntu 16 | 
| Platform                                 | Intel NUC |
| Language                            | Python               | 

### Issue Description
As I am using D435 Depth Camera , I am facing problem regarding wait_for_frames function .  
 
frames = pipeline.wait_for_frames()

While waiting for frame ; sometimes takes more than 5000 ms ( 8s to 2 minutes ) of time to take the frame .During that time execution of code is stuck and as the time limit exceeds default timeout limit , though any kind of exception is not occurring . Is it possible that wait_for_frames() stuck the whole code, irrespective of it ( whole script ) is running on thread ?

ev-mp, 2019-08-22:
@TanmayPathak , it is unreasonable that any SDK thread will lock beyond 15 seconds without raising an exception. 
Can you reproduce the deadlock with realsense-provided demos/tools?
Can you provide kernel/application/console logs for review ?

RealSenseCustomerSupport, 2019-09-10:
----------------------------------------------

Hi @TanmayPathak,

Please let us know whether or not have you managed to reproduce the issue.

Thank you!

TanmayPathak, 2019-09-10:
No, I am not able to reproduce that error till now. I will let you know if
it happens again.

RealSenseCustomerSupport, 2019-09-10:
----------------------------------------------

Please open new issue whenever you will be able to reproduce it and provide us
with above mentioned logs and information.

Thank you!

closed 2019-09-10 by RealSenseCustomerSupport