realsenseai/librealsense #4672: Is there any API to get the status(busy or not) of realsense camera?

xhuan28, 2019-08-20:
I have more than one realsense camera connected. I would like to query devices and enable the one that is not in use. Is there any API for that?

dorodnic, 2019-08-20:
Hi @xhuan28 
There is no such generic API. You could try to start streaming an catch the exception if it already does. Or you could manage this in your own app (by updating a file, registry or some other method).
Key issues is that there is no universal API across Windows and Linux that lets us know if camera is in use. There are most likely hardware specific tricks that would work for one family of devices but not others. What kind of RealSense devices are you using?

xhuan28, 2019-08-20:
I have one D435 and one T265. If there is no generic API for me to know if they are in use or not, perhaps I can access it by serial number. 
However, I met another issue besides that. I want to access them in a separate process respectively, but even in that scenario, there is still an issue.
I create a process to access D435 first and then create another process to access T265, but error occurs.

> 16:48:18.575 [11536] [E] Device-BDA0: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
16:48:18.586 [11536] [E] Device-BDA0: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
terminate called after throwing an instance of 'rs2::backend_error'
  what():  xioctl(VIDIOC_S_FMT) failed Last Error: Device or resource busy

dorodnic, 2019-08-20:
Yes. In case of T265 it can only be accessed from a single process (we are going to fix that).
For the D400 you could access it from different processes but you cannot start streaming twice.

xhuan28, 2019-08-21:
Ok, I see. May I know which release it is going to be fixed?

RealSenseSupport, 2020-09-11:
Thank you for highlighting this issue and need for a new API for T265. We have moved our focus to our next generation of products and consequently, we will not be addressing this issue in the T265.

closed 2020-09-16 by RealSenseSupport