realsenseai/librealsense #4676: error with USB driver

JackStig, 2019-08-20:
| | |
|---|---|
|**librealsense**|2.25.0 RELEASE|
|**OS**|Linux|

When I update the USB driver,link between realsense and my virtrue machine smashed.After that ,I can't use realsense-viewer .I'm new to realsense,any help will be appreciated.

dorodnic, 2019-08-20:
Hi @JackStig 
What do you mean by updating the usb driver?

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

JackStig, 2019-08-20:
In a terminal I wrote  `realsense-viewer` and the GUI started.After 3 seconds there is a text on top-right which suggest me to update something about USB,meanwhile keep realsense in touch.But link between realsense and virtual machine smashed.
Now I've learnt that there is something to be noticed if I want to use realsense in a virtual machine from https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md. I'm going to install realsense again in a new virtual machine.

JackStig, 2019-08-20:
By the way I'm using VM Workstation Pro15

dorodnic, 2019-08-20:
You are probably referring to firmware update notification? If it is the case, the camera will get disconnected couple of times during this process. Not sure how it should behave within a VM, but if you have to use the camera in a VM, just ignore the update recommendation or update once on a different PC.

JackStig, 2019-08-21:
Thank you for the reply,now I fix my problem.

closed 2019-08-27 by dorodnic