realsenseai/librealsense #4681: T265 not enumerating correctly after hard reboot

jmachuca77, 2019-08-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.0.18.5715 | 
| Operating System & Version |   Nvidia Jetpack 4.2.1  | 
| Kernel Version (Linux Only)    |  4.9.140-tegra                                       | 
| Platform                                 | Nvidia Xavier/Nano/TX2  |
| SDK Version                            |  2.25.0                          | 
| Language                            |  N/A                          | 
| Segment			|  N/A                          | 

### Issue Description
When booting up a system the T265 is enumerated on the USB bus as 03e7:2150 which I believe is the Movidius Chip, so when trying to use the camera with any script like rs-enumerate-devices or rs-pose I get an error:
 ```
() $ rs-enumerate-devices 
No device detected. Is it plugged in?

() $ rs-pose
RealSense error calling rs2_pipeline_start_with_config(pipe:0x558d0bffe0, config:0x558d09f5e0):
    No device connected
```

This is the sequence of events after boot:
Running lsusb, rs-enumerate-devices and then lsusb again.
```
() $ lsusb
Bus 002 Device 002: ID 2109:0813 VIA Labs, Inc. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 2dae:1011  
Bus 001 Device 003: ID 03e7:2150  
Bus 001 Device 002: ID 2109:2813 VIA Labs, Inc. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

() $ rs-enumerate-devices 
No device detected. Is it plugged in?

() $ lsusb
Bus 002 Device 002: ID 2109:0813 VIA Labs, Inc. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 2dae:1011  
Bus 001 Device 003: ID 03e7:2150  
Bus 001 Device 002: ID 2109:2813 VIA Labs, Inc. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

After unplugging and replugging the camera I get the correct behavior and everything works as expected

This the output from lsusb after replugging the camera, then rs-enumerate-devices, and then lsusb again:
```
() $ lsusb
Bus 002 Device 002: ID 2109:0813 VIA Labs, Inc. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 2dae:1011  
Bus 001 Device 005: ID 03e7:2150  
Bus 001 Device 002: ID 2109:2813 VIA Labs, Inc. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

() $ rs-enumerate-devices 
Device info: 
    Name                          : 	Intel RealSense T265
    Serial Number                 : 	845412110624
    Firmware Version              : 	0.0.18.5715
    Physical Port                 : 	vid_8087 pid_0B37 bus_2 port_1
    Product Id                    : 	0B37
    Product Line                  : 	T200

Stream Profiles supported by Tracking Module
 Supported modes:
    stream       resolution      fps       format   
    Fisheye 1	  848x800	@ 30Hz	   Y8
[... 1057 lines cut ...]
[   17.328377] fuse init (API version 7.26)
[   18.432397] wlan0: authenticate with f0:9f:c2:25:f9:c9
[   18.439909] wlan0: send auth to f0:9f:c2:25:f9:c9 (try 1/3)
[   18.444293] wlan0: authenticated
[   18.446837] wlan0: associate with f0:9f:c2:25:f9:c9 (try 1/3)
[   18.451455] wlan0: RX AssocResp from f0:9f:c2:25:f9:c9 (capab=0x411 status=0 aid=1)
[   18.453847] wlan0: associated
[   18.457525] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  148.021150] usb 1-2.1: USB disconnect, device number 3
[  155.168833] usb 1-2.1: new high-speed USB device number 5 using tegra-xusb
[  155.191314] usb 1-2.1: New USB device found, idVendor=03e7, idProduct=2150
[  155.191327] usb 1-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  155.191332] usb 1-2.1: Product: Movidius MA2X5X
[  155.191336] usb 1-2.1: Manufacturer: Movidius Ltd.
[  155.191340] usb 1-2.1: SerialNumber: 03e72150
[  165.684438] usb 1-2.1: USB disconnect, device number 5
[  165.905137] usb usb2: usb_suspend_both: status -16
[  165.924122] usb usb2: usb_suspend_both: status 0
[  166.193646] usb 2-1.1: new SuperSpeed USB device number 3 using tegra-xusb
[  166.214490] usb 2-1.1: New USB device found, idVendor=8087, idProduct=0b37
[  166.214524] usb 2-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  166.214657] usb 2-1.1: Product: Intel(R) RealSense(TM) Tracking Camera T265
[  166.214661] usb 2-1.1: Manufacturer: Intel(R) Corporation
[  166.214665] usb 2-1.1: SerialNumber: 8454121106240000
```

MartyG-RealSense, 2019-08-20:
I was not quite sure of the sequence of events from your description.  From the title about a hard reboot, it sounds as though the camera works the first time that the computer is started up, but is no longer detected after a reboot of the computer.  Is that what is happening, please?

If it is, it sounds similar to an effect reported in a couple of cases with 400 Series cameras, where the camera stops working after the computer's reboot but is corrected with an unplug-replug of the USB cable.  It was believed that the problem in those cases may be related to an issue with the computer's BIOS configuration.

Below are links to the reported cases that I know of:

https://github.com/IntelRealSense/librealsense/issues/1283

https://github.com/IntelRealSense/librealsense/issues/1615

jmachuca77, 2019-08-20:
Sorry for the confusing name. But no this happens after any type of boot, be it initial boot (hard boot) or a soft reboot. Always the camera is detected as the Movidius processor without a way to change it to be the camera. Only after power cycling the camera (unplug usb and replug) then it can be accessed as a camera.

MartyG-RealSense, 2019-08-20:
Thanks for the clarification.   I have seen a couple of past cases where the T265 has been ID'd as Movidius.  Here's one of them:

https://github.com/IntelRealSense/librealsense/issues/3436

jmachuca77, 2019-08-20:
Yes, I have gone through those issues. But running rs-pose does not work as the camera is not detected. And I am running the latest release built from source on the jetson platform with the options
```-DBUILD_PYTHON_BINDINGS=bool:true -DPYTHON_EXECUTABLE=/usr/bin/python -DBUILD_EXAMPLES=true -DBUILD_WITH_CUDA=true -DBUILD_WITH_TM2=ON```
Which should include support for the t265 and they do as replugging the camera fixes the issue but coming from boot it does not work and unplugging the camera is not an option as it will be inside a housing.

MartyG-RealSense, 2019-08-20:
It looks as though you have covered most of the possibilities that I can think of.  There have been a couple of Jetson users who had different errors that were solved by trying a different USB cable.  

https://github.com/IntelRealSense/librealsense/issues/4578

I apologize that I'm unable to provide more specific solutions, as my knowledge of the deeper workings of the T265 model is limited.  One of the Intel support guys on this forum will likely be able to advise you well.

jmachuca77, 2019-08-20:
I have tried with different cables but the one that comes with the camera seems to be the best one to use (this is the one I am using) I am going through a powered USB3 hub. However I cannot connect the camera directly to the Xavier because when I do that I get disconnected from the camera randomly, I think the port on the Xavier cannot give enough power to keep the camera alive.
I will try that again later today. And I'll report back on the findings here. Thank you.

MartyG-RealSense, 2019-08-20:
USB power issues have also been known with the RealSense 400 Series camera models on some single-board computing devices such as Up Board and Intel Compute Stick.  An Intel support representative once said that the camera needs 2W of power reliably supplied to it, which was why a powered USB 3 hub was ideal because it could supply up to 12W and so could meet the camera's power needs if the power supply was unreliable when the camera was inserted directly into a USB port on the board.

I would have thought that a top-end board such as the Xavier would be less prone to such issues, but hardware often finds a way to defy expectations.   :)    Good luck!

jmachuca77, 2019-08-22:
Tried with different USB cables and even connecting the camera directly to the USB port on the Xavier and got exactly the same issue.

MartyG-RealSense, 2019-08-22:
I am wondering if it is a physical issue with the USB port on the Xavier.  With my own computer, a Windows machine, I sometimes hear a repeating USB disconnect and reconnect sound effect from my mouse even if I am not touching it at the time (though my leg may brush against the cable slightly)  My mouse will go dead and its lights go out, sometimes accompanied by an on-screen warning message, and comes back to life when I give its USB connector a gentle push into the port.   This happens with the ports on another PC too.

MartyG-RealSense, 2019-08-27:
Is your NetMate hub one that plugs into the mains electricity please, or is it a "passive" hub that draws its power from the PC?

MartyG-RealSense, 2019-08-27:
When a RealSense model has apparent USB related issues with an Up Board, a mains powered USB hub has a better chance of correcting the problem than a passive hub, which cannot provide the stable wattage that a mains hub can.  I understand how it impacts mobility though.

MartyG-RealSense, 2019-08-27:
The original Up Board only has a 'USB 3.0 OTG' micro port (not a full size USB 3 port) and a USB 2.0 port.  The newer UP Squared has a full USB 3.0 port.  Is your Up model an Up Squared, please?

MartyG-RealSense, 2019-08-27:
Ok, thank you very much.

Some further digging brought to light that the T265 has had issues earlier this year with rs-enumerate-devices not detecting the T265.  

https://github.com/IntelRealSense/librealsense/issues/3465

Are you able to get the T265 to work in the RealSense Viewer even if rs-enumerate-devices said it is not detected?

MartyG-RealSense, 2019-08-27:
I apologize that the diagnostic process is taking some time and I appreciate all the information that you and others with this problem are patiently providing.

Since the T265 can function after unplug-replug, it is presumably not a mechanical fault with the OTG port or the hubs that is preventing initial detection on boot.

As mentioned earlier in this conversation thread, it is thought that an issue related to the BIOS configuration of a particular computer may cause the detection problem on first boot-up.  I recall a case where somebody tried it with three PCs, and it didn't work with two of them but did with the third.

https://github.com/IntelRealSense/librealsense/issues/4681#issuecomment-523135157

I'll refer this case back to Intel member @ev-mp who had some initial contact with this case 6 days ago through labelling it as a T265 related issue, though they may not be able to respond until tomorrow.

MartyG-RealSense, 2019-08-28:
@radfordi   Thanks so much for the information!

@rdinoff. @jmachuca77   I am not an Intel employee, so are not on the inside loop on internal discussions.  So I rely on the brains of kind fellows like @radfordi to fill in my knowledge gaps.  :)

BriceRenaudeau, 2019-08-28:
Thanks for the information,
I also have the same problem

doisyg, 2019-08-30:
> @rdinoff, this is a known hardware issue. We are trying to work with Movidius to find a workaround. Currently, if the Movidius processor get voltage before the hub itself, then it fails to detect the USB speed correctly.

Is there a rough estimate of when the problem will be fixed? Days, weeks or months? We would like to use the t265 on some production system and we need to make a decision soon.

patrickpoirier51, 2019-09-02:
Following... Same Issue on my mini powered hub

Sequence from cold start
[    5.185719] hub 1-1:1.0: 4 ports detected
[    5.519724] usb 1-1.1: new high-speed USB device number 3 using ehci-platform
[    5.689179] usb 1-1.1: config 1 interface 0 altsetting 0 bulk endpoint 0x81 has invalid maxpacket 64
[    5.707505] usb 1-1.1: config 1 interface 0 altsetting 0 bulk endpoint 0x1 has invalid maxpacket 64
[    5.740710] usb 1-1.1: New USB device found, idVendor=03e7, idProduct=2150, bcdDevice= 0.01
[    5.757399] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    5.772002] usb 1-1.1: Product: Movidius MA2X5X
[    5.781053] usb 1-1.1: Manufacturer: Movidius Ltd.
[    5.790624] usb 1-1.1: SerialNumber: 03e72150

disconnect reconnects
[  496.769441] usb 1-1.1: USB disconnect, device number 3
[  502.624214] usb 1-1.1: new high-speed USB device number 4 using ehci-platform
[  502.795307] usb 1-1.1: New USB device found, idVendor=03e7, idProduct=2150, bcdDevice= 0.01
[  502.795324] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  502.795330] usb 1-1.1: Product: Movidius MA2X5X
[  502.795335] usb 1-1.1: Manufacturer: Movidius Ltd.
[  502.795340] usb 1-1.1: SerialNumber: 03e72150
[  506.633665] usb 1-1.1: USB disconnect, device number 4
[  507.133848] usb 1-1.1: new high-speed USB device number 5 using ehci-platform
[  507.285287] usb 1-1.1: New USB device found, idVendor=8087, idProduct=0b37, bcdDevice=ff.ff
[  507.285298] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  507.285304] usb 1-1.1: Product: Intel(R) RealSense(TM) Tracking Camera T265
[  507.285310] usb 1-1.1: Manufacturer: Intel(R) Corporation
[  507.285315] usb 1-1.1: SerialNumber: 9053121115070000

patrickpoirier51, 2019-09-02:
Well... that might be a short term solution:
uhubctl is utility to control USB power per-port on smart USB hubs. Smart hub is defined as one that implements per-port power switching.
https://github.com/mvp/uhubctl

patrickpoirier51, 2019-09-05:
@radfordi  Jim, here is my '' Plan B '' to get the T265 flying without having to disconnect-reconnect 
- Home made USB cable with power routing to an external cable
- Relay control of the power with the NC contact on T265 power
- Driven by BananaPi GPIO (opto-Driver)
- Shell script to Power reset before launching LibRealsense

[image]

Please note the Masking Tape Skirt to reduce direct sunlight exposure during  outdoor tests

skylouis, 2019-09-26:
hi @jmachuca77 I have the same issue, how did you get around this problem ?

jmachuca77, 2019-09-26:
didn't, its still an issue.

skylouis, 2019-09-27:
hi @ev-mp when is Intel planning to patch this issue ?

RealSenseCustomerSupport, 2019-10-04:
----------------------------------------------

At this point in time, the workaround is to make sure that the host platform or USB hub has power and ready for enumeration prior to power provided to the T265.

Datasheet Statement (https://dev.intelrealsense.com/docs/tracking-camera-t265-datasheet):
If T261/T265 is being connected to host system via USB hub, keep in mind that T261/T265 will enter into USB
enumeration protocol as soon as 5V has been provided on VBUS pin. USB protocol is handled by the Movidius MA215x
device. Make sure USB hub is connected and enumerated to host system prior to power being supplied to T261/T265.

closed 2019-10-11 by RealSenseCustomerSupport

jmachuca77, 2019-10-11:
Great! I'll just ship the product with a midget that makes sure to plug the camera in after the host system is up! thanks Intel!

doisyg, 2019-10-12:
Why closing the issue? Does this mean this problem will not be solved? This make it hard to use in a production ready system

MrG4dget, 2019-10-14:
I have to agree with @doisyg ! Its a real pain to get the T265 going! 
Having to hack a cable, and having scripts to power on/of the camera is just not a solution.

skylouis, 2019-10-14:
Same for us, @RealSenseCustomerSupport  you have now released the T261 for production environment but this issue make it really difficult for us to integrate it in our product. We have done complex HW modification to accommodate this lack of flexibility for Intel. We hope that a FW patch  will solve this issue soon !

zwn, 2020-02-08:
@RealSenseCustomerSupport Can we have this issue open while waiting for a solution? The problem can be experienced on multiple platforms as already reported here by many people and elsewhere on the internet. We are currently experiencing the issue on APU2 board. The suggested workaround is of no help since we cannot control when 5V is introduced to different parts of the board. T265 is alone with this problem.

> @rdinoff, this is a known hardware issue. We are trying to work with Movidius to find a workaround. Currently, if the Movidius processor gets voltage before the hub itself, then it fails to detect the USB speed correctly.

@rdinoff Is any solution to this on the horizon?

zwn, 2020-02-08:
> @rdinoff Is any solution to this on the horizon?

Err sorry, I meant to call out @radfordi.

skylouis, 2020-02-08:
If you want I can share with you the hardware solution we have put in place to control the 5V alimentation using GPIO of the single board computer.

fnoop, 2020-03-18:
@RealSenseCustomerSupport - Please, don't be stupid.  This is a USB sensor most commonly connected to some embedded/robotic device, there is every reasonable expectation that it will be plugged in at power/boot time and expected to work.  This is a fault in your product, pure and simple.  This is a recall level fault, if you can't fix it in firmware.

rmackay9, 2020-03-18:
I can confirm that with an RPI4 at least, @patrickpoirier51's suggestion of using uhubctrl to power cycle the USB hub does work.  I've commented on this issue as well: https://github.com/IntelRealSense/librealsense/issues/5527

radfordi, 2020-03-18:
@fnoop, the T265 comes with a [1yr warranty](https://store.intelrealsense.com/buy-intel-realsense-tracking-camera-t265.html).  Please contact [customer support](https://www.intel.com/content/www/us/en/support.html) if you'd like to return your camera.

@zwn, the only solution we've found is to run `uhubctrl`.  We've considered doing the power cycle ourselves in `librealsense` before trying to boot the camera to avoid needing a separate program, but even then, you would still need to have power control of the USB port for this to work.

cowboygary, 2020-06-26:
Hi all, what's actually the real solution?

dexterpengji, 2020-07-28:
Same issue: T265 will not be detected if it was connected before booting the computer. I have to replug it to the computer after booting to let it work.
Hope this issue could be solved soon.

ethanrublee, 2020-07-31:
TL;DR;  soft reboots of the nano cause enumeration to fail.  It can reliably be fixed using the command ``uhubctl -l 1-2 -p 1 -a cycle`` if there is no hub plugged into the nano, and the t265 is plugged into ports 2-4, but not 1.

[image]

For the jetson nano developer kit, i was having inconsistent results with @radfordi's suggestion of using ``uhubctrl``.

It seems like ``uhubctrl`` only effectively turns off the power for all dev kit USB ports on the nano if its issued for the first port on the 2-1 bus with ``./uhubctl -l 1-2 -p 1 -a cycle``  Also, I noticed that if I plugged the t265 into the 1st port, it would sometimes enumerate and sometimes not, indicating the race condition, with an error like:

```
$ rs-enumerate-devices 
 30/07 23:00:01,517 ERROR [547566735376] (tm-boot.h:39) Error booting T265
 30/07 23:00:04,571 ERROR [547566735376] (tm-boot.h:39) Error booting T265
No device detected. Is it plugged in?
```
I also get the above error if I soft reboot the nano with ``sudo reboot``, regardless of the port i plug into.

However, plugging the t265 into any other port (2-4), and then cycling port 1 with uhubctl, i have repeatable behavior.

The error also occurs if the the t265 is plugged into another port, and that port is cycled with uhubctl.  I assume this is because the port does not actually loose its power.

Listing hub and port state with uhubctl
```
$ ./uhubctl
Current status for hub 2-1 [0bda:0411 Generic 4-Port USB 3.1 Hub, USB 3.10, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-2 [0bda:5411 Generic 4-Port USB 2.1 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0503 power highspeed enable connect [03e7:2150 Movidius Ltd. Movidius MA2X5X 03e72150]
  Port 4: 0103 power enable connect [1d50:606f canable.io canable gs_usb 001D00385852430320363530]
```

Turn off port 1:
```
$ ./uhubctl -l 1-2 -p 1 -a off
Current status for hub 2-1 [0bda:0411 Generic 4-Port USB 3.1 Hub, USB 3.10, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
Sent power off request
New status for hub 2-1 [0bda:0411 Generic 4-Port USB 3.1 Hub, USB 3.10, 4 ports]
  Port 1: 00a0 off
Current status for hub 1-2 [0bda:5411 Generic 4-Port USB 2.1 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
Sent power off request
New status for hub 1-2 [0bda:5411 Generic 4-Port USB 2.1 Hub, USB 2.10, 4 ports]
  Port 1: 0000 off

```
List it again, notice the USB devices have disconnected.
```
$ ./uhubctl
Current status for hub 2-1 [0bda:0411 Generic 4-Port USB 3.1 Hub, USB 3.10, 4 ports]
  Port 1: 00a0 off
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-2 [0bda:5411 Generic 4-Port USB 2.1 Hub, USB 2.10, 4 ports]
  Port 1: 0000 off
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
```

Turn back on
```
$ # ./uhubctl -l 1-2 -p 1 -a on
Current status for hub 2-1 [0bda:0411 Generic 4-Port USB 3.1 Hub, USB 3.10, 4 ports]
  Port 1: 00a0 off
[... 39 lines cut ...]
[16529.668747] usb 1-2: usb_suspend_both: status 0
[16531.316800] usb 2-1: usb_suspend_both: status 0
[16531.468778] usb 2-1: usb_suspend_both: status 0
[16531.469056] tegra-xusb-padctl 7009f000.xusb_padctl: power on UTMI pads 1
[16531.469169] usb usb2: usb_suspend_both: status 0
[16531.608749] tegra-xusb-padctl 7009f000.xusb_padctl: power down UTMI pad 1
[16531.628701] usb 1-2: usb_suspend_both: status 0
[16531.825162] tegra-xusb-padctl 7009f000.xusb_padctl: power on UTMI pads 1
[16532.032728] usb 1-2.3: new high-speed USB device number 88 using tegra-xusb
[16532.054736] usb 1-2.3: New USB device found, idVendor=03e7, idProduct=2150
[16532.054753] usb 1-2.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[16532.054765] usb 1-2.3: Product: Movidius MA2X5X
[16532.054777] usb 1-2.3: Manufacturer: Movidius Ltd.
[16532.054787] usb 1-2.3: SerialNumber: 03e72150
[16532.352688] usb 1-2.4: new full-speed USB device number 89 using tegra-xusb
[16532.375323] usb 1-2.4: New USB device found, idVendor=1d50, idProduct=606f
[16532.375336] usb 1-2.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[16532.375346] usb 1-2.4: Product: canable gs_usb
[16532.375354] usb 1-2.4: Manufacturer: canable.io
[16532.375362] usb 1-2.4: SerialNumber: 001D00385852430320363530
[16532.377118] gs_usb 1-2.4:1.0: Configuring for 1 interfaces
[16533.116000] IPv6: ADDRCONF(NETDEV_CHANGE): can0: link becomes ready
```

Enumeration and disabling the power fails if there is a hub plugged into a port on the nano (haven't been able to test with one supported by ``uhubctl``), see this ticket https://github.com/mvp/uhubctl/issues/258

AllenIsaacJose, 2023-09-15:
SOLVED

```
sudo sh -c "echo 141a0000.pcie > /sys/bus/platform/drivers/tegra194-pcie/unbind"
sudo sh -c "echo 141a0000.pcie > /sys/bus/platform/drivers/tegra194-pcie/bind"
```
Find the PCIe card to which the USB is connected. In my case `141a0000.pcie`. The above command will turn the power supply to the PCIe card off and then on thereby making my T265 tracker camera detectable.