realsenseai/librealsense #4682: Extract ROI from depth frame

LossOfHumanIdentity, 2019-08-20:
Hi, 

How do you extract Region Of Interest (ROI) from a depth frame? 

My goal is simply to multiply the depth frame for a bi dimensional binary mask (basically a simple matrix with ones in the Region Of Interest and zeros everywhere else) to, in the end, generate a pointcloud only in a certain region of interest. 

I can do this simply with basic webcams and OpenCV, but how do you perform the above mentioned simple and important task with RealSense? 

Any help will be higly appreciated

MartyG-RealSense, 2019-08-20:
The RealSense SDK 2.0 software does have a means of setting a ROI.  In the RealSense Viewer software, it can be done by defining a 'setpoint' value in the **Advanced Controls > AE Control** section of the Viewer's side-panel options.  Page 2 of Intel's camera tuning guide, in section E, describes this process.

https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/BKMs_Tuning_RealSense_D4xx_Cam.pdf

I believe that a way to define a value for this setpoint outside of the RealSense Viewer software is to set it in a camera configuration file called a json, also known as a Visual Preset.

https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets

A json can be loaded into the RealSense Viewer or loaded into your own program via code.

You could define this json file yourself, though an easier way to create a custom preset file is to define the settings in the RealSense Viewer and then export the settings as a json using a save icon on the toolbar at the top of the Viewer's options side-panel.  The setpoint command is listed in the json as **aux-param-autoexposure-setpoint**

The Visual Presets page has a list of links to json configuration files so that you can see how they are structured.

https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets#preset-table

LossOfHumanIdentity, 2019-08-20:
Could you provide some reference on how to this with the Python or C++ API?

I can not use Software Tools, I have to integrate this in a bigger Python or C++ based pipeline.

Thank you so much in advance.

MartyG-RealSense, 2019-08-20:
I just found such a Python reference whilst checking for further reading for you.

https://github.com/IntelRealSense/librealsense/issues/2681#issuecomment-436391721

LossOfHumanIdentity, 2019-08-20:
Thank you a lot for your assitance, but I guess I have not been clear enough with what I'm asking. 

RealSense SDK enables me to get the pointcloud of a given depth_frame
```
depth_frame = frames.get_depth_frame()
points = pc.calculate(depth_frame)
```
What I am _trying_ to do, is simply get the correspondent `depth_image` from the depth_frame and multiply it for a binary bi dimensional `mask` of same size (e.g 640x480, a simple matrix made of ones and zeros) in order to discard the depth information everywhere except where the bi dimensional mask value is 1.
`depth_image = depth_image * mask`

My _only_ goal now is to understand how to compute the pointcloud not with the original depth_frame, but with the just obtained above mentioned depth_image.

Is this anyhow possible? I am confident of performing such operations with simple webcams and OpenCV but am really strugglying to understand how to actually get this simple task to work with RealSense. 

Thanks again in advance, I hope this can be clarified

MartyG-RealSense, 2019-08-20:
The scope of your project exceeds my level of RealSense programming knowledge, unfortunately, though one of the Intel team members on this forum should hopefully be able to provide some suggestions for how to approach it.   I wish you the best of luck!

ev-mp, 2019-08-21:
@LossOfHumanIdentity , in Librealsense the frames are intentionally designed to be _immutable_ to allow safe multi-thread sharing and usage, and in general in-place modifications of frame data should be avoided.

The ways to handle this scenario in SDK are:
- Implement a custom processing block (e.g `stencil mask`) that will produce a _modified copy_ of the original frame. This will extend the code base and allow other users to benefit from this as well, but will require code modification on multiple levels, including API changes. This is the way that the  SDK filters - `align`, `colorizer`, `post-processing` and others are implemented. This however requires 
- Using a `Software Device` class ([example link](https://github.com/IntelRealSense/librealsense/blob/master/examples/software-device/rs-software-device.cpp)):
   - Create and configure `Software Device` instance
   - Get a copy the raw depth data from the actual frame
   - Modify the depth data locally
   - Feed the software device with the modified data
   - Pass the "software-generated" depth frame to the SDK's `pointcloud` processing block

I would recommend that you try the `software device` method.
Both ways a functionally equivalent, though the latter is much faster to prototype and also does not require SDK changes.

LossOfHumanIdentity, 2019-08-21:
Hi @ev-mp, thank you for your reply, even though I'm not sure I fully understood how to actually use the Software Device you mentioned for this specific scenario. 

Anyway, I've been able to perform the above mentioned task in a Python environment by simply getting the depth_image, masking it as desired and then call the native OpenCV `cv2.reprojectImageTo3D` method like stated in the following: 

`points = cv2.reprojectImageTo3D(disparity, Q)`

I am currently using the following 
`Q = np.float32([[1, 0, 0, -0.5 * 640],[0, 1, 0, -0.5 * 480],[0, 0, 0, 383],[0, 0, 1, 0]])
`

But I would like to know if you could please provide an accurate Q matrix for D435i?

[image]

Thank you so much in advance, I guess that if you provide me the correct Q matrix I can close this issue and mark it as solved!

ev-mp, 2019-08-22:
The SDK provides methods to gather the required components for the openCv's `Q` matrix: 
- Sensor intrinsic data ([example](https://github.com/IntelRealSense/librealsense/blob/master/examples/measure/rs-measure.cpp#L283-L285))
- Inter-sensor extrinsic and baseline data ([example](https://github.com/IntelRealSense/librealsense/blob/master/tools/depth-quality/depth-quality-model.cpp#L812-L813))
- The distortion coefficients are zeroed as the raw depth is already undistorted and rectified, so it saves you the extra-handling.

The terminology and units are mostly the same so this should be a straight-forward mapping.

Note however, that Realsense depth frame provides **_Depth (Z)_** values while the mentioned `cv2.reprojectImageTo3D` takes **_Disparity_** as input. I couldn't find a direct openCv equivalent for the SDK's `rs2_deproject_pixel_to_point`.
So in order to generate 3D data with openCv you'll have first to transform the Depth data to the Disparity domain.

The SDK provides a method to generate a `scaled disparity` data required by the post-processing filters ([example](https://github.com/IntelRealSense/librealsense/tree/master/examples/post-processing)), so it give you a starting point. #3344, #3740.

closed 2019-08-23 by LossOfHumanIdentity