realsenseai/librealsense #4689: D435 freezes when enabling depth frame

zinan93, 2019-08-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version | Linux (Ubuntu 16/17) | 
| Kernel Version (Linux Only)    |             4.15.0-58-generic                           | 
| Platform                                 | PC  |
| SDK Version                            |     2.25.0                      | 
| Language                            |  python/ROS                   | 
| Segment			|  Robot            | 

### Issue Description
I'm using SDK 2.25 with realsense D435. When enabling depth stream using realsense-viewer the whole window freezes, giving the following error in the terminal:

```
ERROR [140156057155328] (synthetic-stream.cpp:44) Exception was thrown during user processing callback!
```

I tried the default mode, adjusting to the lowest resolution and frame rate for depth stream, and it still always freezes. I have a quite powerful desktop PC with USB 3.0 port. I tried reinstalling the SDK and noticed that when installing `librealsense2-dkms` I got the following error:

```
Error! Bad return status for module build on kernel: 4.15.0-58-generic (x86_64)
Consult /var/lib/dkms/librealsense2-dkms/1.3.5/build/make.log for more information.
```

and by going into `/var/lib/dkms/librealsense2-dkms/1.3.5/build/make.log`:

```
DKMS make.log for librealsense2-dkms-1.3.5 for kernel 4.15.0-58-generic (x86_64)
2019年 08月 21日 星期三 12:24:19 CEST
make: Entering directory '/usr/src/linux-headers-4.15.0-58-generic'
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/uvc_driver.o
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/uvc_queue.o
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/uvc_v4l2.o
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/uvc_video.o
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/uvc_ctrl.o
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/uvc_status.o
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/uvc_isight.o
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/uvc_debugfs.o
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/uvc_entity.o
  LD [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/uvcvideo.o
  MODPOST 1 modules
  CC      /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/uvcvideo.mod.o
  LD [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/uvcvideo.ko
make: Leaving directory '/usr/src/linux-headers-4.15.0-58-generic'
make: Entering directory '/usr/src/linux-headers-4.15.0-58-generic'
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/v4l2-core/v4l2-dev.o
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/v4l2-core/v4l2-ioctl.o
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/v4l2-core/v4l2-device.o
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/v4l2-core/v4l2-fh.o
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/v4l2-core/v4l2-event.o
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/v4l2-core/v4l2-ctrls.o
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/v4l2-core/v4l2-subdev.o
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/v4l2-core/v4l2-clk.o
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/v4l2-core/v4l2-async.o
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/v4l2-core/v4l2-compat-ioctl32.o
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/v4l2-core/vb2-trace.o
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/v4l2-core/v4l2-trace.o
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/v4l2-core/v4l2-mc.o
[... 314 lines cut ...]
    Color	  320x240	@ 6Hz	   BGRA8
    Color	  320x240	@ 6Hz	   RGBA8
    Color	  320x240	@ 6Hz	   BGR8
    Color	  320x240	@ 6Hz	   YUYV
    Color	  320x180	@ 60Hz	   RGB8
    Color	  320x180	@ 60Hz	   Y16
    Color	  320x180	@ 60Hz	   BGRA8
    Color	  320x180	@ 60Hz	   RGBA8
    Color	  320x180	@ 60Hz	   BGR8
    Color	  320x180	@ 60Hz	   YUYV
    Color	  320x180	@ 30Hz	   RGB8
    Color	  320x180	@ 30Hz	   Y16
    Color	  320x180	@ 30Hz	   BGRA8
    Color	  320x180	@ 30Hz	   RGBA8
    Color	  320x180	@ 30Hz	   BGR8
    Color	  320x180	@ 30Hz	   YUYV
    Color	  320x180	@ 6Hz	   RGB8
    Color	  320x180	@ 6Hz	   Y16
    Color	  320x180	@ 6Hz	   BGRA8
    Color	  320x180	@ 6Hz	   RGBA8
    Color	  320x180	@ 6Hz	   BGR8
    Color	  320x180	@ 6Hz	   YUYV
```

Any help is appreciated! Thx!

ev-mp, 2019-08-21:
@zinan93 , Do you manage to run `rs-capture` and `rs-depth` demos correctly?
For the dkms-related issue please follow #4669.

zinan93, 2019-08-21:
just tried rs-capture and rs-depth, also doesn't work. But installing the WA DKMS works. Many thx! I shall keep an eye on the debian fix then.

ev-mp, 2019-08-21:
@zinan93 , please provide the following data for review:
- Kernel log `dmesg -wT`
- Console logs for `rs-depth` and `rs-capture`
- Librealsense log: 
     -   run `realsense-viewer` -> click on configuration in the upper right corner -> change the log file verbosity level to "DEBUG" -> clock "Apply" -> take a note of the log file name/path
    -  rerun `realsense-viewer` and try to stream depth and color stream
   -  attach and post the librealsense log

zinan93, 2019-08-22:
Yeah so I tested today with the manually installed WA DKMS SDK 2.25 is not stable, it still sometimes freezes. I updated to SDK 2.26 and it's the same. 

* So for kernel log please see [here](https://github.com/zinan93/realsense_log/blob/926226bec88c3bb1b87b09509395f88fd71be09e/dmesglog.txt#L1)
* now both rs-depth and rs-capture runs smoothly most of the time, but sometimes it freezes and there's not terminal output
* realsense-viewer log see [here](https://raw.githubusercontent.com/zinan93/realsense_log/master/librealsense.log)

RealSenseCustomerSupport, 2019-09-25:
----------------------------------------------

Hi @zinan93,

The issue is currently under investigation.Updates will be posted as soon as investigation findings become available.

Thank you!

RealSenseCustomerSupport, 2020-02-28:
----------------------------------------------

Hi @zinan93,

Do you experience the same issue using the latest build of RealSense SDK?

Thank you!

RealSenseCustomerSupport, 2020-03-05:
----------------------------------------------

@zinan93,

Do you still need help with this question?

zinan93, 2020-03-05:
Hi, thank you for the reminder. Unfortunately my old project was concluded and I no long have access to my lab PC and camera. Let's close this for now.

closed 2020-03-05 by zinan93