realsenseai/librealsense #4695: Is there a way to know D435's parameters?

Tangpearl, 2019-08-21:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win 10 / Linux (Ubuntu 16 /  |                                       
| Platform                                 | PC| 
| Language                            |  C++  | 

### Issue Description
I want to know my D435 camera's status, what can I do? It seems that intel realsense viewer can't do it.

MartyG-RealSense, 2019-08-21:
The SDK has a tool called **sensor-control** that can give information about a camera and the sensors supported on that particular model.  

If you are using Windows, you can find a pre-built executable version of the tool by downloading and installing the full SDK, and then going to the SDK folder location below:

**C: > Program Files (x86) > Intel RealSense SDK 2.0 > Tools > rs-sensor-control**

If you are using Linux, or prefer to build the tool from source-code instead of using a pre-built version, you can find the code here:

https://github.com/IntelRealSense/librealsense/tree/master/examples/sensor-control

Tangpearl, 2019-08-21:
thanks very much

Tangpearl, 2019-08-21:
> The SDK has a tool called **sensor-control** that can give information about a camera and the sensors supported on that particular model.
>
> If you are using Windows, you can find a pre-built [...]

Tangpearl, 2019-08-21:
so do you know how to change these parameters through codes?

MartyG-RealSense, 2019-08-21:
The 'API How To' page (the version of it with text explanations) is a good place to start.  If you have follow-up questions after that about specific functions, you are welcome to add comments here on this discussion.

https://github.com/IntelRealSense/librealsense/wiki/API-How-To#get-first-realsense-device

closed 2019-08-27 by dorodnic