realsenseai/librealsense #4699: Error about post-processing

Tangpearl, 2019-08-21:
I'm  doing some work about D435 post-processing. I want to change the filtered rs2::frame into cv::Mat and do some following work. However, there happens an error. To be convenient, I wrote the part where the problem happens into a test.cpp. However, the error is beyond me. Hope to get some help. Appreciate it very much.

----------------------------------------------------------------------------------------------------
//g++ -std=c++11 test.cpp -o test `pkg-config --libs opencv4` `pkg-config --cflags opencv4` `pkg-config --libs realsense2` `pkg-config --cflags realsense2` -lpthread
#include <librealsense2/rs.hpp> // Include RealSense Cross Platform API
#include <opencv2/opencv.hpp>

#include <iostream>
#include <fstream>
#include <vector>
#include <cmath>
#include <unistd.h>
#include <mutex>
#include <thread>
#include <atomic>
#include <chrono>
#include <time.h>

static const int w = 640;
//! Height
static const int h = 480;
///@}

static const int d_fps = 30; //!< Frame rate for D435

static const double D435_MIN = 0.11; // | - min and max depths of D435 (m)
static const double D435_MAX = 2.00; // |

static const std::string DEPTH_SNO = "819612073628"; //!< Serial number of D435
static const std::string TRACK_SNO = "909212111324"; //!< Serial number of T265
#define BUFFER_LENGTH 5 // length of buffer for cameras

#define INPUT_RATE 50 // Rate at which camera feed is taken (Hz)       | - Maximum rates

#define MAP_UPDATE_RATE 10 // Rate at which global map is updated (Hz) |
class Camera {
public:
	rs2::context ctx;

	std::vector<rs2::pipeline> pipelines; // pipelines for depth and tracking cameras - 0: tracking, 1: depth
   
	float scale; //!< Depth scale (m)
	//@{
	//! Focal length: x (pixels)
	float fx;
	//! Focal length: y (pixels)
	float fy;
	//@}
	//@{
	//! Image center: x (pixels)
	float ppx;
	//! Image center: y (pixels)
	float ppy;
	//@}
	//! Distortion model type
	int model;
	//! Distortion Coefficients
	float coeffs[5];
	///@}
   
	//! Queue for Depth frames
	rs2::frame_queue d_queue;
	/*! Queue for Pose frames */
	rs2::frame_queue t_queue;

};
 void Update (Camera const &C, cv::Mat const &depth) {
	    /*quaternion q_T265 (pose.rotation.x, pose.rotation.y, pose.rotation.z, pose.rotation.w);
	    quaternion t_T265 (pose.translation.x, pose.translation.y, pose.translation.z, 0);
	    quaternion q_G_D435 = q_T265 * Q_T265_D435 * quaternion(1,0,0,0);
	    quaternion t_G_D435 = t_T265 + q_T265 * T_T265_D435 * q_T265.inv();
	    quaternion pose_pix (0, 0, 0, 0);*/
            int count=0;
	    float x_D435, y_D435, z_D435;
            //z_D435 = depth.at<unsigned short int>(0,0) * C.scale;
            //std::cout<<z_D435<<std::endl;
	    for (int i = 0; i < h; i++) {
	        for (int j = 0; j < w; j++) {
                    count++;
                    //std::cout<<count<<std::endl;
	            z_D435 = depth.at<unsigned short int>(i,j) * C.scale;
                    //std::cout<<z_D435<<std::endl;
	            x_D435 = (j-C.ppx)/C.fx * z_D435;
	            y_D435 = (i-C.ppy)/C.fy * z_D435;

	            //pose_pix = t_G_D435 + q_G_D435 * quaternion(x_D435,y_D435,z_D435,0) * q_G_D435.inv();

	            //if (z_D435 > D435_MIN && z_D435 < D435_MAX)
	                //g_map->update_point (pose_pix.x, pose_pix.y, pose_pix.z);
	        }
	    }
	}
int main() try
{
    int num_dev = 0;
    int num_frame = 0;
    std::atomic_bool alive {true};
    Camera C;
    for (auto&& dev : C.ctx.query_devices())
	    {
 		num_dev++;
             }
    std::cout<<"There are "<<num_dev<<"cameras connect"<<std::endl;
    if(num_dev == 0) return 0;
    rs2::pipeline pipe(C.ctx);
    rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_DEPTH, w, h, RS2_FORMAT_Z16, d_fps);
    std::cout << "Depth Camera initialized: {" << w << "," << h << "}, 90 FPS\n";
    rs2::pipeline_profile profile = pipe.start(cfg);
    rs2::decimation_filter dec_filter;
    rs2::threshold_filter thr_filter;
    rs2::spatial_filter spat_filter;    // Spatial    - edge-preserving spatial smoothing
    rs2::temporal_filter temp_filter;
    rs2::disparity_transform depth_to_disparity(true);
    rs2::disparity_transform disparity_to_depth(false);
    
    dec_filter.set_option(RS2_OPTION_FILTER_MAGNITUDE, 3);
    thr_filter.set_option(RS2_OPTION_MIN_DISTANCE, 0.1);       
    thr_filter.set_option( RS2_OPTION_MAX_DISTANCE, 4.0);
       std::thread exit_check([&]() {
        while (alive) {
            if (std::cin.get() == ' ') {
                alive = false;
            }
        }
    });
     while (alive) {
       rs2::frameset data = pipe.wait_for_frames();
       rs2::frame depth_frame = data.get_depth_frame();
       //num_frame = depth_frame.get_frame_number();
       //std::cout<<num_frame<<std::endl;
       rs2::frame filtered = depth_frame;
       // Note the concatenation of output/input frame to build up a chain
       filtered = dec_filter.process(filtered);
       filtered = thr_filter.process(filtered);
       //filtered = depth_to_disparity.process(filtered);
       //filtered = spat_filter.process(filtered);
       //filtered = temp_filter.process(filtered);
       //filtered = disparity_to_depth.process(filtered);
       if(!filtered) std::cout<<1;
       cv::Mat depth(cv::Size(w, h), CV_16UC1, (void *)filtered.get_data(), cv::Mat::AUTO_STEP);
       //int a=depth.cols;int b=depth.rows;
       //std::cout<<a<<" "<<b<<" ";
       Update(C, depth);
       
}
      return 0;
}
    	catch (const rs2::error & e) {
		std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
		}
	catch (const std::exception & e) {
		std::cerr << e.what() << std::endl;
};

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Operating System & Version |  Linux  Ubuntu 16   | 
| Platform                                 | PC |
| Language                            |  C++                          |

Tangpearl, 2019-08-21:
Just now I find that as long as I remove the decimation filter, the code can run. I'm wondering why this can happen.

ev-mp, 2019-08-21:
@Tangpearl , the decimation filter [changes the depth frame size](https://github.com/IntelRealSense/librealsense/blob/development/doc/post-processing-filters.md#decimation-filter), as well as the frame intrinsic.
When iteration over the filtered depth data adjustments shall be made for the new dimension and intrinsic to avoid buffer overruns and misalignment
>float x_D435, y_D435, z_D435;
            //z_D435 = depth.at<unsigned short int>(0,0) * C.scale;
            //std::cout<<z_D435<<std::endl;
	    for (int i = 0; i < ~~**h**~~; i++) {
	        for (int j = 0; j < ~~**w**~~; j++) {
                    count++;
                    //std::cout<<count<<std::endl;
	            z_D435 = depth.at<unsigned short int>(i,j) * C.scale;
                    //std::cout<<z_D435<<std::endl;
	            x_D435 = (j-C.ppx)/~~**C.fx**~~ * z_D435;
	            y_D435 = (i-C.ppy)/~~**C.fy**~~ * z_D435;

#2958, #2802

dorodnic, 2019-08-27:
Hi @Tangpearl 
Did this help?

Tangpearl, 2019-09-10:
Yeah, it did.

Tangpearl, 2019-09-10:
Thanks very much.@ev-map

Tangpearl, 2019-09-10:
Thanks very much, @ev-mp

ev-mp, 2019-09-10:
@Tangpearl , thanks for the confirmation.

closed 2019-09-10 by ev-mp