realsenseai/librealsense #4700: T265 fails to start with 2.26.0

JBBee, 2019-08-21:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.1.0.242 | 
| Operating System & Version |   Win 10 | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | PC |
| SDK Version                            |  2.26.0                        | 
| Language                            | Viewer                         | 
| Segment			|  AR                        | 

### Issue Description
Steps to reproduce in the viewer:
- Run the T265 around an area and then stop it and export the localization map 
- Import the map and start the T265 and run it for a few seconds, then stop it
- Import the map again and try to start the T265 and it fails to start.  You have to unplug the module and re-plug it back in to get it to work again

This is a new issue with 2.26.0

Thanks.

radfordi, 2019-08-22:
Thanks for the quick feedback, @JBBee.  I'll try and reproduce this.

JBBee, 2019-08-29:
Thanks @radfordi This works great now.  I can update our app to 2.27.0 now.  We load the localization map each time an AR session is started so we needed this.

Thanks again!

closed 2019-08-31 by dorodnic