realsenseai/librealsense #4701: D415, Android, Java Example, Dropped Frames

medlraffy, 2019-08-22:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.11.11.100 | 
| Operating System & Version |   Android, 8.0.0  |  
| Platform                                 | Samsung Galaxy S8  |
| SDK Version                            |  2.26.0  | 
| Language                            |  Java    | 
| Segment			| Smartphone | 

### Issue Description

I'm running the demo [/wrappers/android/examples/java_example.](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android/examples/java_example)

I have the camera connected with an USB-C to USB-C OTG cable.

I get about 5-20 frames and then get spammed by the message `I/librs: Dropped frame. alloc_frame(...) returned nullptr` until eventually `waitForFrames()` throws an "UNKNOWN_ERROR".

I tried to supply `Pipeline` with a `Config` with only the depth data enabled but the result was the same.
```
Config config = new Config();
config.enableStream(StreamType.DEPTH, 640, 480);
Pipeline pipe = new Pipeline();
pipe.start(config);
```

Here is the tail of my log:
```
2019-08-21 16:48:32.841 8082-8937/com.example.realsense_java_example I/librs: Dropped frame. alloc_frame(...) returned nullptr
2019-08-21 16:48:32.863 8082-8937/com.example.realsense_java_example I/librs: Dropped frame. alloc_frame(...) returned nullptr
2019-08-21 16:48:32.875 8082-8937/com.example.realsense_java_example I/librs: Dropped frame. alloc_frame(...) returned nullptr
2019-08-21 16:48:32.897 8082-8937/com.example.realsense_java_example I/librs: Dropped frame. alloc_frame(...) returned nullptr
2019-08-21 16:48:32.909 8082-8937/com.example.realsense_java_example I/librs: Dropped frame. alloc_frame(...) returned nullptr
2019-08-21 16:48:32.919 8082-8937/com.example.realsense_java_example I/librs: Dropped frame. alloc_frame(...) returned nullptr
2019-08-21 16:48:32.929 8082-8937/com.example.realsense_java_example I/librs: Dropped frame. alloc_frame(...) returned nullptr
2019-08-21 16:48:32.940 8082-8937/com.example.realsense_java_example I/librs: Dropped frame. alloc_frame(...) returned nullptr
2019-08-21 16:48:32.952 8082-8937/com.example.realsense_java_example I/librs: Dropped frame. alloc_frame(...) returned nullptr
2019-08-21 16:48:32.956 8082-8926/com.example.realsense_java_example I/System.out: frameset try-catch: java.lang.RuntimeException: UNKNOWN_ERROR
2019-08-21 16:48:32.962 8082-8937/com.example.realsense_java_example W/librs: Frame received with streaming inactive,Depth0, Arrived,0.000000 1566431312962.790039
2019-08-21 16:48:33.674 8082-8936/com.example.realsense_java_example I/librs: read endpoint: 130, active requests count: 0
2019-08-21 16:48:33.674 8082-8936/com.example.realsense_java_example I/librs: USB pipe 130 reset successfully
2019-08-21 16:48:33.674 8082-8936/com.example.realsense_java_example I/librs: start join stream_thread
2019-08-21 16:48:33.675 8082-8936/com.example.realsense_java_example I/librs: Transfer thread stopped for endpoint address: 130
2019-08-21 16:48:33.675 8082-8926/com.example.realsense_java_example I/librs: The user was holding on to 16 frames after stream 0x0x7793c0d9e0 stopped
```

medlraffy, 2019-08-22:
It streams correctly if I revert to an older dependency:
```
dependencies {
    implementation 'com.intel.realsense:librealsense:2.19.0@aar'
}
```

RealSenseCustomerSupport, 2019-09-17:
----------------------------------------------

@medlraffy https://github.com/IntelRealSense/librealsense/pull/4731 should address the issue. Could you please try the latest SDK v2.28 to see if you still get issue? Thanks!

RealSenseCustomerSupport, 2019-09-24:
----------------------------------------------

@medlraffy Could you please update if SDK v2.28 fix your issue? Thanks!

RealSenseCustomerSupport, 2019-10-12:
----------------------------------------------

@medlraffy Didn't get response from your side for week.s Will close it at this point. Please create new issues if you still have questions/issues. Thanks!

closed 2019-10-12 by RealSenseCustomerSupport