realsenseai/librealsense #4702: Camera time stamp question

tonyhsiao1995, 2019-08-22:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version |   Ubuntu16.04  | 
| Kernel Version (Linux Only)    | 4.15.0-55-generic  | 
| Platform                                 | PC |
| SDK Version                            |    2.23.00   | 
| Language                            |  C++  | 

### Issue Description
1. I am trying to make sure that the D435 gets the newest frameset which timestamps should close to the system clock.
The camera is Global time and queue_size is 16 now. 
I am using 30 fps and 1280x720 resolution. However, the time_diff will have a range from 15 - 50 ms. Did I do anything wrong? I assume that the time_diff should <= 1000/30
I used rs-data-collect to check the stable of frame rate and it looks good, 31-34 ms per frame.
```
while (true) {
    rs2::pipeline pipeline_ = devices[device_id]->get_camera_pipeline();
    rs2::frameset frameset;
    unsigned int timeout_realsense = 40;
    std::chrono::milliseconds unix_time;
    bool success = pipeline_.try_wait_for_frames(&frameset, timeout_realsense);
    unix_time = 
        std::chrono::duration_cast<std::chrono::milliseconds>(
            std::chrono::system_clock::now().time_since_epoch());

    if (!success) {
      continue;
    }
    double frame_timestamp = frameset.get_timestamp();
    double time_diff = fabs(frame_timestamp - unix_time.count());
    std::cout << time_diff << std::endl;
  }
```

2. How to set the camera time domain to RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME

ev-mp, 2019-08-22:
@tonyhsiao1995 , it is not clear what _streams_ are are configured in this use-case.
In case you have both RGB and Depth streaming then the framesets will include both. So then the question is whether he intra-stream sync is active, because if not then the frames will be be produced asynchronously.
To invoke the internal sync (RGB will be the master) you have to configure identical FPS, enforce `FPS over Exposure priority` . Follow #3685, #774 for details.

The timestamp domain is predetermined by the underlying OS capabilities (metadata) and cannot be controlled directly #4525. There are ways to enforce it but I would suggest to review the use-case as using the Host clock is less accurate than using the Hardware/Global time clocks due to the obvious kernel-imposed latency

tonyhsiao1995, 2019-08-22:
OK, Thanks for the reply. 
Another question for the time stamp, I am writing my function.
```
bool RealsenseBridge::GetLatestFrameset(rs2::frameset *frameset) {
  int success = pipeline_.try_wait_for_frames(frameset);
  if (success) {
    try {
      auto unix_time = std::chrono::duration_cast<std::chrono::milliseconds>(
          std::chrono::system_clock::now().time_since_epoch());
      std::cout << std::setprecision(15);
      std::cout << "Frameset time = " << frameset->get_timestamp() << std::endl;
      std::cout << "Epoch time = " << unix_time.count() << std::endl;
    }
    catch (...) {
      std::cerr << "GetLatestFrameset failed\n"; 
    }
  }
  return success;
}
```
I have two programs to call this function. One program is like a unit test, only have a  while loop to call the function repeatedly and it works very well. However, in the real application program, I got the wrong timestamp like below. The frameset time is old. This error will happen at approximate 10% of calling the function. Is there any scenario will cause the wrong timestamp?
```
Frameset time = 1566473895069.6
Epoch time = 1566474198987
Ros time = 1566474198.987119178
```

tonyhsiao1995, 2019-08-23:
I solve the problem.
The program executes making folder. It will cause camera time problem
```
std::string mkpath = "mkdir -p " + log_path;
system(mkpath.c_str())
``` 
I close the issue

closed 2019-08-23 by tonyhsiao1995