realsenseai/librealsense #4703: Can some help to reduce noise in depth measurements

martijnkusters, 2019-08-22:
D415
sdk 2.25.0
firmware updated.

Currently we are trying to reduce the depth noise measurement at about 1.4 m distance.
In the figure below I show the result of 1 pixel measurement.

[image]

 The distance varies in time. I use already all filters. 
Can some help what parameters I can tweak to reduce the variation?

Best Regards,
Martijn Kusters

MartyG-RealSense, 2019-08-22:
Although depth noise increases over distance (a phenomenon called RMS error), it only becomes particularly noticeable when observing surfaces further than around 3 meters from the camera.

You may be able to get a further reduction in noise by turning off the IR Emitter component, which projects a semi-random pattern of dots onto the scene to help the camera to read details from flat surfaces such as walls and doors.  The laser-based pattern projector in the camera can create a noise effect called 'laser speckle' on the image.  

Alternatives to turning off the IR Projector are to reduce the visibility of the pattern by reducing the value of the Laser Power option, or excluding the pattern using a Visual Preset filter for the D415 called **'Left Imager Color w/o IR Pattern'**.  

https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets#preset-table

If you require a dot pattern projection, you could use an external LED based projector which should help to remove the laser speckle phenomenon.

Intel's camera turning guide - which is an excellent reference guide in general for improving camera images - states: 

"When using a laser-based projector, it is extremely important to take all means necessary to reduce the laser-speckle or it will adversely affect the depth. 
 Non-coherent light sources will lead to best results in terms of depth noise, but cannot normally generate as high contrast pattern at long range. Even when all speckle reduction methods are employed it is not uncommon to see a passive target or LED projector give >30% better depth than a laser-based projector". 

https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/BKMs_Tuning_RealSense_D4xx_Cam.pdf

kingbaggot, 2019-08-22:
yeah - I've updated the firmware and am 2m from my flat surface and the a focussed point can jump around 80 mm which is throwing a lot of stuff off for me. 

I've set the camera to the 'Left Imager Color w/o IR Pattern' and have no flourescents on in the room, and I don't get any noticeable change. 

If you find a way to reduce this I would LOVE to know.

agrunnet, 2019-08-22:
In your data it ranges from 1386 to 1395. This is 11mm peak to peak or probably 3.5mm RMS. 

A single point is expected to have an RMS error of about 2.8mm at this distance so not far off, depending a bit on the pattern it is looking at. For example if you turn off laser and use random Ly textured surface you can probably improve this 2x.   
Also try to turn down the exposure a bit. (Yes this is counter intuitive but it helps). 
Finally, the more you temporally average the better it gets. 

I would say that

kingbaggot, 2019-08-22:
hi agrunnet - If I turn off the laser then I get no readings at all - is that correct ?

MartyG-RealSense, 2019-08-22:
The IR projector is useful for adding readable depth detail to flat / low texture objects in a scene but the camera does not rely exclusively on it.  It can also process depth using the ambient light in the area.

RealSenseCustomerSupport, 2019-09-11:
----------------------------------------------

Can this ticket be closed if no further questions?

closed 2019-09-20 by RealSenseCustomerSupport