realsenseai/librealsense #4707: T265 Camera Pose data in Unreal Engine

martinthelwell, 2019-08-22:
Hi, 
The Unreal Engine plugin recognises the T265 camera, 
and I see that 6dof data is listed as a format in the plugin code.

Is there currently a way to access the 6dof pose data within a Blueprint using the plugin?

Thanks,

RealSenseCustomerSupport, 2019-09-13:
----------------------------------------------

Hi @martinthelwell - T265 is not fully supported in Unreal at this time.  You may get the pose data and the device recognized but full support is not available at this time.

We will keep this in mind for future.

Thanks

closed 2019-09-13 by RealSenseCustomerSupport

maciejcybinski, 2019-10-17:
Hi, what do you mean by "You may get the pose data"? Is it just a matter of exposing the c++ code to the blueprint or it requires writing the reader?

kromond, 2019-12-07:
Full support for all the data streams is not essential, just pose data would be great.  My C++ skills are not developed enough to get this without some clues.  Could someone help?

aksharif05, 2020-02-12:
T265 support now in Unreal

aksharif05, 2020-03-05:
intel realsense tracking camera t265 output value  save and live update in CSV file?

YaoDaDi666, 2021-02-22:
how can i get pose from T265 and send to UE4???

vickyhexr, 2021-03-09:
@YaoDaDi666  you can create a python application for T265 Pose data and send it to unreal with the help of socket programming.

VitalyZaikin, 2021-03-18:
> @YaoDaDi666 you can create a python application for T265 Pose data and send it to unreal with the help of socket programming.

Of course it's possible. Can you show an example. Woul be helpfull