realsenseai/librealsense #4708: CustomRW can't open device (Debian 9)

belugum, 2019-08-22:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version |   Debian 9 | 
| Kernel Version (Linux Only)    |  4.19.0-5 | 
| Platform                                 | PC |
| SDK Version                            | 2.6.8.0 | 
| Language                            |  CLI | 
| Segment			| Other | 

### Issue Description
Hi

I'm trying to read the device calibration data of a Intel RealSense D435 using Intel.Realsense.CustomRW. The device is detected but I can't access the data:
```
$ Intel.Realsense.CustomRW -list
CustomRW for Intel RealSense D400, Version: 2.6.8.0

Device Name                                  Serial Number       Firmware Version    
Intel RealSense D435                         829212070383        05.11.06.250        
$ sudo Intel.Realsense.CustomRW -r -sn 829212070383
CustomRW for Intel RealSense D400, Version: 2.6.8.0

Error: cannot find or open device with serial number 829212070383
```

The camera is well detected on the viewer and all the other utilities I tried.
CustomRW work perfectly on Windows 10 (same hardware). Any idea on what could solve this?

Thanks!

MartyG-RealSense, 2019-08-22:
If the camera works normally except for calibration, I wonder if the calibration table inside the camera might be corrupted.  A 'gold reset' to factory defaults of the camera with **CustomRW -g** may correct this if the table is corrupted.

belugum, 2019-08-23:
Same thing for `Intel.Realsense.CustomRW -g`: `Error: cannot find or open device with serial number 829212070383`

The same command works fine on Windows 10 (same computer, same camera), so it's not an hardware issue. 

My kernel version is not supported for the librealsense2-dkms package so that probably causes this. I found instructions to patch the kernel manually for Ubuntu with other kernel versions but I'm on Debian and it doesn't work. Is there a workaround?

MartyG-RealSense, 2019-08-23:
Kernel patching can be bypassed with the CMake customization flag **-DFORCE_LIBUVC=true**

belugum, 2019-08-27:
I reinstalled and double-checked the udev-rules, but CustomRW still doesn't work.

RealSenseCustomerSupport, 2019-09-24:
----------------------------------------------

Hi @belugum,

Link to the Debian packages for RealSense D400 Series Calibration Tools along with
installation instructions can be found here:
https://downloadcenter.intel.com/download/27415/Intel-RealSense-D400-Series-Dynamic-Calibration-Software-Tools

Please let us know if this has solved the issue.

Thank you!

RealSenseCustomerSupport, 2019-10-22:
----------------------------------------------

Hi @belugum,

Do you need any further clarifications regarding this topic?

closed 2019-10-29 by RealSenseCustomerSupport