realsenseai/librealsense #4710: odroid xu4 realsense d415 Frame didn't arrived within 5000 after 30 seconds

nk2IsHere, 2019-08-22:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   |  05.11.06.250  | 
| Operating System & Version |  Linux Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.14.133-166  | 
| Platform                                 |  Odroid XU4  |
| SDK Version                            |   2.<?>.<?>  | 
| Language                            |  python  | 
| Segment			|  others  | 

### Issue Description
<Describe your issue / question / feature request / etc..>

RealSenseCustomerSupport, 2019-09-17:
----------------------------------------------

@nk2IsHere Which example did you run when get this issue? Thanks!

nk2IsHere, 2019-09-17:
I ran https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_viewer_example.py
rs-depth
realsense viewer

I’ve found temporary solution: changed the usb cable to usb 2.0 to type-c. This fixed random crashes, though downgraded the speed.

RealSenseCustomerSupport, 2019-09-29:
----------------------------------------------

@nk2IsHere Did you try the installation according to https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_odroid.md

nk2IsHere, 2019-09-29:
Yes, I did

RealSenseCustomerSupport, 2019-11-22:
----------------------------------------------

@nk2IsHere Did you get this issue with the original USB3 cable for the camera?

nk2IsHere, 2019-11-22:
Yes, I got this issue while using original usb 3.0 cable

RealSenseCustomerSupport, 2020-03-06:
----------------------------------------------

@nk2IsHere Did you get chance to see if you still have such issue with the latest FW and SDK? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-03-27:
----------------------------------------------

@nk2IsHere Could you please update if you still need support on this? Thanks!

nk2IsHere, 2020-03-27:
Thank you, but I don’t need a support for this. Buying a new type-c to usb 3.0 cable is a fix of this problem.

closed 2020-03-27 by nk2IsHere