realsenseai/librealsense #4711: How to create a desktop app with something like Electron

lamroger, 2019-08-23:
I've been able to run the Node examples locally and want to build a desktop app with librealsense. node-librealsense is very easy to use for me and I'd like to use something like Electron to build an app for other people to use. 

I don't have experience with develping desktop apps, especially cross-platform. Does anyone know if it would work or how it would work?

There seem to be dependencies for each architecture that need to be installed. Would I request each user to download those prereqs first? Would they need to build their own librealsense and would that be in an install step?

Any guidance would be appreciated!

I'm using a D415 on a Mac.

RealSenseCustomerSupport, 2019-12-05:
----------------------------------------------

@lamroger Sorry for late response. Seems this issue is out of scope of librealsense. Did you get it through now?  Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-01-15:
----------------------------------------------

@lamroger Will close this if no other questions. Thanks!

lamroger, 2020-01-15:
Sounds good. Thank you!

closed 2020-01-15 by RealSenseCustomerSupport