realsenseai/librealsense #4712: Can't open the deploy.prototext or deploy.caffe model

jawad-tech, 2019-08-23:
/caffe/caffe_io.cpp:1121: error: (-2:Unspecified error) FAILED: fs.is_open(). Can't open "MobileNetSSD_deploy.prototxt" in function 'ReadProtoFromTextFile'

I dont know why? Please help. I have used the same application from realsense wrapper and mad a ros node for it. The program "catkin_make" successfully build it but it provides error when i run the node. it says that the file cannot be opened as the errir stated above.

I have been to this post https://github.com/IntelRealSense/librealsense/issues/3173 but not understand how to set the directory?

RealSenseCustomerSupport, 2019-10-05:
----------------------------------------------

Hi @jawad-tech,

You can set the directory using datadir variable.

Thank you!

closed 2019-10-22 by RealSenseCustomerSupport

JaeYoungLee79, 2022-01-10:
sd
Hi@RealSenseCustomerSupport 

You can set the directory using datadir variable.
>>> I don't understand.

Directory? datadir variable?

Would you more explanation?

mamoonik, 2022-09-01:
Please answer the question @RealSenseCustomerSupport