realsenseai/librealsense #4713: Multiple D435 devices RGB cameras flicker from light to black intermittently

didory123, 2019-08-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D400 } | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version |   {Win 10 |                                 | 
| Platform                                 | PC  |
| SDK Version                            |  2.25 }                          | 
| Language                            |  {C++ }                          | 

### Issue Description
Hi guys,

I currently have 2 D435 cameras setup at the same time, working on a multi-camera application. I am running into this weird problem where when I have the 2 devices running at the same time, the RGB frames for both cameras flicker from light to dark intermittently. 

Has anybody seen this happen before? If so, I would be extremely grateful to learn how to fix this issue! 

Thank you.

MartyG-RealSense, 2019-08-23:
If you are doing a depth to color sync with your D435's, this may cause problems in a multi-camera setup according to the advice given in the link below.

https://github.com/IntelRealSense/librealsense/issues/2637#issuecomment-434203140

didory123, 2019-09-17:
Hi guys! 
I'm sorry, I completely forgot to comment and close this issue.
My issue was indeed the master-slave connection on the RGB connection. My application doesn't require that yet, so I turned it off for now. Thank you for the help and the useful links though!

closed 2019-09-17 by didory123