realsenseai/librealsense #4714: Simple and important questions for me，please help me, thank you very much!

carryyu, 2019-08-23:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {ubuntu16.04  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

It's my code.For example, if I want to process this frame, the processing time of this frame is about 1 second. After processing this frame, the next frame is processed, but the next frame taken at this time is the next frame after one second. All the frames during the 1 second are lost, what should I do?
 I hope that it will not take frames when I process the current frame, so the next frame I get is the next frame of the current frame instead of the frame after one second.
```
pipeline = rs.pipeline()
config = rs.config()
rs.config.enable_device_from_file(config, bag_file,repeat_playback=False)
profile = pipeline.start(config)
align_to = rs.stream.color
align = rs.align(align_to)
frame_id = 1
while True:
    frames = pipeline.wait_for_frames()
    for frame in frames:
        print frame_id
        print frame
        aligned_frame = align.process(frames)  # align depth frame to color frame
        aligned_depth_frame = aligned_frame.get_depth_frame()
        color_frame = aligned_frame.get_color_frame()
        if not aligned_depth_frame or not color_frame:
            continue
        depth_image = np.asanyarray(aligned_depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())
        depth_color_image = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)
        cv2.imwrite('/home/lzy/lzy/bagData/20190731_154005/testcolor/' + str(frame_id) + '.jpg', color_image[:, :, ::-1])
        cv2.imwrite('/home/lzy/lzy/bagData/20190731_154005/testdepth/' + str(frame_id) + '.jpg', depth_color_image)
        frame_id+=1
    time.sleep(1)
```

carryyu, 2019-08-23:
It's the Code

```
import pyrealsense2 as rs
import cv2
import numpy as np
import time
import thread
num = '005'
bag_file = '/home/lzy/lzy/bagData/20190731_154'+num+'.bag'

pipeline = rs.pipeline()
config = rs.config()
rs.config.enable_device_from_file(config, bag_file,repeat_playback=False)
profile = pipeline.start(config)
align_to = rs.stream.color
align = rs.align(align_to)
# quene = rs.frame_queue(10)
frame_id = 1

while True:
    frames = pipeline.wait_for_frames()
    aligned_frame = align.process(frames)  # align depth frame to color frame
    aligned_depth_frame = aligned_frame.get_depth_frame()
    color_frame = aligned_frame.get_color_frame()

    if not aligned_depth_frame or not color_frame:
        continue

    depth_image = np.asanyarray(aligned_depth_frame.get_data())
    color_image = np.asanyarray(color_frame.get_data())
    depth_color_image = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)
    cv2.imwrite('/home/lzy/lzy/bagData/20190731_154005/testcolor/' + str(frame_id) + '.jpg', color_image[:, :, ::-1])
    cv2.imwrite('/home/lzy/lzy/bagData/20190731_154005/testdepth/' + str(frame_id) + '.jpg', depth_color_image)
    frame_id+=1
    time.sleep(1)

```

carryyu, 2019-08-23:
According to the code, if I want to process this frame, the processing time of this frame is about 1 second(time.sleep(1)). After processing this frame, the next frame is processed, but the next frame taken at this time is the next frame after one second. All the frames during the 1 second are lost, what should I do?
I hope that it will not take frames when I process the current frame, so the next frame I get is the next frame of the current frame instead of the frame after one second.

g2-bernotas, 2019-08-23:
What happens if you tell it to play not in real-time:

``` Python 
profile = pipeline.start(config)
playback=profile.get_device().as_playback()
playback.set_real_time(False)
```

carryyu, 2019-08-24:
> What happens if you tell it to play not in real-time:
> profile = pipeline.start(config)
> playback=profile.get_device().as_playback()
> playback.set_real_time(False)
Thank you very much, it works!

closed 2019-08-24 by carryyu