realsenseai/librealsense #4716: Android Align Frames

erikgallegos, 2019-08-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Android 8.0  | 
| Platform                                 | Android |
| SDK Version                            |  2.25                         | 
| Language                            |  kotlin                       | 
| Segment			| Smartphone    | 

### Issue Description
I am trying to use the `Align` class to align my depth frame to the color stream but I'm not sure exactly how to use it or what it's supposed to do. I've tried both aligning and not aligning my depth frame but I don't see a noticeable change in the output.

This is how I'm currently using it:

```
val align = Align(StreamType.COLOR)
val frame = align.process(depthFrame) as DepthFrame
```

Is this the correct way to use this? What kind of behavior should I expect? Does it matter that I am casting it to a DepthFrame after? Any help would be appreciated.

Thank you.

matkatz, 2019-09-03:
Aligning a depth frame to a color frame creates a new depth frame with the same resolution as the color stream in which each depth pixel matches the color pixel on the color frame at the same index.
The opposite direction will create a new color frame with matching pixels.
You should call process/applyFilter on a frame set and not on a single frame.

For example:
```
private Align mAlign = new Align(StreamType.DEPTH);
...
try(FrameSet frames = mPipeline.waitForFrames()) {
    try(FrameSet p = frameSet.applyFilter(mAlign)) {
        try( Frame d = p.first(StreamType.DEPTH, StreamFormat.Z16)) {
            DepthFrame depath = d.as(Extension.DEPTH_FRAME);
            //do something with depth
        }
        try( Frame c = p.first(StreamType.COLOR, StreamFormat.RGB8) ){
            VideoFrame color = c.as(Extension.VIDEO_FRAME);
            //do something with color
        }    
    }
}
```
An output of the code above will look like:
[image]

Please notice that smaller field of view of the color sensor cause a cropped color image.

erikgallegos, 2020-04-14:
@matkatz in the above example, what would be the `Frame` / `FrameSet` used to upload into the `GLRsSurfaceView`?

RealSenseCustomerSupport, 2020-04-29:
----------------------------------------------

Hi @erikgallegos,

I think you can read the code.
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/examples/capture/src/main/java/com/intel/realsense/capture/MainActivity.java#L149

RealSenseCustomerSupport, 2020-05-11:
----------------------------------------------

@erikgallegos Any other questions about this? Looking forward to your update. Thanks!

closed 2020-05-18 by erikgallegos