realsenseai/librealsense #4718: How to smooth exported ply

visus100, 2019-08-24:
| requierd info |  |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | D400_Series_FW_5_11_11_100 | 
| Operating System & Version |   Win (8.1/10)  | 
| Platform                                 | PC |
| SDK Version                            |  SDK-WIN10-2.23.0.909                       | 
| Language                            |  C++                        | 

### Question

After exporting point cloud the surface looks like pixels sharp edges. Does anyone have idea how make (programmatically) to surface smooth (with some kind of examples would be great) of course i mean not implementing smooth shader but geometries change.

I was trying to do some approaches with post-processing filters but it looks like it is the part of exporting thing (or maybe i'm wrong).

### actual result

[image]

### expectation result

[image]

MartyG-RealSense, 2019-08-24:
The links below may provide useful advice about smoothing out a point cloud.

https://github.com/IntelRealSense/librealsense/issues/1053

https://github.com/IntelRealSense/librealsense/issues/1375

When exporting a ply, also take care to set the depth units correctly.

https://github.com/IntelRealSense/librealsense/issues/2009

Some users have reported that they get better results from exporting the ply from the RealSense Viewer program instead of a script such as export_to_ply.

visus100, 2019-08-24:
Thanks for fast reply.

I digged a lot about it already and i did a lot of things to improve quality of scan. For example: change distance, take care of light conditions, changing power of emitter, etc.. 

Of course i can get a little bit better result than this what i show but still is looks like "minecraft blocks" (its just for compare)

I was wondering if someone could help me with adding somehow algorithms called "Mesh Smoothing" or "Mesh denoiser" before exporting it to ply.

Here example (paper) of  algoritmes what i'm trying to reach:
https://isaacguan.github.io/2018/01/26/Mesh-Smoothing/

Here the demo of upper example:
https://isaacguan.github.io/projects/mesh-smoothing/

Other example (paper):
https://wang-ps.github.io/denoising.html

agrunnet, 2019-08-24:
Please tell us more about the setting you chose. This looks worse that it should be. 
Depth units =100
Distance=20cm
Turn down laser to 60mW
848x480
90fps
Post process with temporal average and down sample 2x. 
Possibly set exposure manually until monchrone image looks quite dark.

visus100, 2019-08-24:
Thanks for sharing this camera settings. This improves my scan expectations (helps me a lot), but still i'm dreaming about some kind of mesh denoiser working just like in example:
https://isaacguan.github.io/projects/mesh-smoothing/

Because it actually looks like points are merged like layers (maybe some kind of improving exporter?)
I'm reaching this smooth model by blender smooth brush in sculpt mode but i need this made programatically.

### actual result

[image]

### expectation result

[image]

### MartyG-RealSense 

> Some users have reported that they get better results from exporting the ply from the RealSense Viewer program instead of a script such as export_to_ply.

Can someone share the code for this trick, couse i would like to compare and can't find this.

I have only 
`points.export_to_ply("filename.ply", color);`

MartyG-RealSense, 2019-08-25:
There is a code example here for export_to_ply:

https://github.com/IntelRealSense/librealsense/issues/2832#issuecomment-443823647

There is also an example program in the Python language called export_ply_example:

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/export_ply_example.py

If you want to programmatically apply processing to a point cloud mesh, it may be better to do it with the Point Cloud Library, which the RealSense SDK has support for via a wrapper.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/pcl

I recommend googling for 'pcl mesh smoothing' for leads on how to do this.

closed 2019-09-17 by visus100

vishnukumarts, 2020-01-10:
> requierd info
> Camera Model 	D435i
> Firmware Version 	D400_Series_FW_5_11_11_100
> Operating System & Version 	Win (8.1/10)
> Platform 	PC
> SDK Version 	SDK-WIN10-2.23.0.909
> Language 	C++
> ### Question [...]

Once I compute the normals  to ply file exported, I am also getting result equivalent to the actual 
result on your 1st post.