realsenseai/librealsense #4719: Question - Using Align proces() in c#

kmeirlaen, 2019-08-25:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.11.11.100 | 
| Operating System & Version |   Windows 10  | 
| Kernel Version (Linux Only)    |                                         | 
| Platform                                 | PC |
| SDK Version                            |  SDK2 }                          | 
| Language                            |  C#                          | 
| Segment			|  windows                          | 

I am trying to modify cs-tutorial-2 example from the SDK to include alignment.
This post (https://github.com/IntelRealSense/librealsense/issues/2734) gave me some directions, leading to an example at https://gist.github.com/ogoshen/58eb3b74b229d8a47d8c6f2bebc099be

However, the above code no longer seems to work.

Looking back, I see the wrapper SDK no longer includes aligning a Frameset, as this code was removed from Processing.cs

`        public FrameSet Process(FrameSet original)
        {
            FrameSet rv;
            using (var singleOriginal = original.AsFrame())
            {
                using (var processed = Process(singleOriginal))
                {
                    rv = FrameSet.FromFrame(processed);
                }
            }
            return rv;
        }`

older version including the snippet:
https://github.com/IntelRealSense/librealsense/blob/3abefc245ba9c2e18e4907299119f0010702a8fa/wrappers/csharp/Intel.RealSense/Processing.cs

I tried adding it, but now I get some other errors.

Can anybody point me into a direction for a working example on how to peform alignment using C#?

Much appreciated.

K.

ogoshen, 2019-08-29:
See [cs-tutorial-3-processing](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/cs-tutorial-3-processing/Window.xaml.cs) sample for current API.

`Align.Process` should still be there but marked as obsolete.

kmeirlaen, 2019-08-29:
Thanks, exactly what I needed!

closed 2019-08-29 by kmeirlaen