realsenseai/librealsense #4722: Playback seek

codesian, 2019-08-25:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | latest | 
| Operating System & Version |   Ubuntu 18  | 
| Kernel Version (Linux Only)    |  4.18)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.24 }                          | 
| Language                            |  {c++ }                          | 
| Segment			|  {others }                          | 

Hi im trying to go to a know bag file time position (60 seconds), im try with playback.seek(nanoseconds) but not works!

`        rs2::config config;
        rs2::context context;
        rs2::frameset frameset;
        
        rs2::playback playback = context.load_device(bag);

        const std::vector<rs2::sensor> sensors = playback.query_sensors();
        for (const rs2::sensor &sensor : sensors) {
            const std::vector<rs2::stream_profile> stream_profiles = sensor.get_stream_profiles();
            for (const rs2::stream_profile &stream_profile : stream_profiles) {
                config.enable_stream(stream_profile.stream_type(), stream_profile.stream_index());
            }
        }

        config.enable_device_from_file(playback.file_name());
        playback.set_real_time( false );
        profile = pipeline.start(config);

        
        while (!end) {
            frameset = pipeline.wait_for_frames();
            if(frameset.size()> 1){
                break;
            }
        }

        using namespace std::chrono_literals;
        std::cout << "DIFF: " << playback.get_position() << std::endl;
        std::cout << "DUR: " << playback.get_duration().count()  << std::endl;
        playback.seek(60000000000ns);
        std::cout << "DIFF2: " << playback.get_position() << std::endl;

        frameset = pipeline.wait_for_frames();

        std::cout << "DIFF: " << frameset.get_color_frame().get_frame_number() << std::endl;
        
        `

Thanks

RealSenseCustomerSupport, 2019-10-16:
----------------------------------------------

Hi @codesian,

Is the issue clarified/resolved? You can also refer to "rs-record-playback" sample code in librealsense and see how it works.

RealSenseCustomerSupport, 2019-12-05:
----------------------------------------------

@codesian,

closed 2019-12-05 by RealSenseCustomerSupport