realsenseai/librealsense #4723: RGB and DEPTH Sync

codesian, 2019-08-25:
| Camera Model                       |  D415            | 
| Firmware Version                   | latest             | 
| Operating System & Version |  Ubuntu 18    | 
| Kernel Version (Linux Only)   |  4.18              | 
| Platform                                 |  PC                |
| SDK Version                           |  2.24             | 
| Language                               |  c++              | 
| Segment			         |  others          | 

Hi we have problems syncing RGB and DEPTH on the same camera, we lower FPS to 6 in order to avoid extra-storage on BAG. Cameras are in movement while record (Human walk speed). Auto exposure is off on RGB camera, Intercam Sync is enabled (we use more cameras). Two sensors not shot at the same time, they do it on diferent moments at specified FPS, but this rate is never aligned between sensors, sometimes 150ms diference, and with movement this artifact is more visible. Is there a way to sync both sensors and get frames at the same time on a single camera? 

Record at high FPS may improve the situation but we have space limits.

Obviusly before write this i read all related issues (#744, #4488...)

Thanks

MartyG-RealSense, 2019-08-26:
The link below summarizes why multi camera color to depth sync can be problematic (because "color sensor always wants to be the master), and discusses other aspects of the subject in detail.

https://github.com/IntelRealSense/librealsense/issues/4493

I would also personally recommend a minimum FPS of 15 if possible, as the experience of past cases shows that going as slow as 6 FPS can introduce problems that disappear at 15 FPS.

codesian, 2019-08-27:
@MartyG-RealSense, thanks for your response, im talking about on single camera rgb-depth sync not on a multicam configuration. 
Im try with 15fps, maybe the situation would be better, just because more frames per second means less diference in distance between sensors shots, but frames remains unsynced.

RealSenseCustomerSupport, 2019-09-16:
----------------------------------------------

Hi

This is the known issue. Currently the sync feature is not implemented for 6FPS, but it works under 30FPS or 15FPS. This support is on current To-Do list but sitting at low priority. Sorry about that.

Thank you

closed 2019-09-28 by RealSenseCustomerSupport