realsenseai/librealsense #4725: MJPEG Color Stream

sam598, 2019-08-26:
The latest release 2.26.0 says that there is support for MJPEG as a stream format:
https://github.com/IntelRealSense/librealsense/pull/4622 - Support for MJPEG stream for selected SKUs

But when trying to enable an MJPEG color stream on a D415 the config fails with:
RuntimeError: Failed to resolve request. No device found that satisfies all requirements

What SKUs support MJPEG streaming? Does it require an unreleased firmware update? Is there something else that needs to be enabled in the config?

dorodnic, 2019-08-26:
Hi @sam598 
At the moment, none of the publicly available devices can benefit from this capability. However, librealsense is shared infrastructure for all supported and future RealSense devices, and we are paving the way for bringing higher-resolutions to RealSense devices. 
You can check what stream profiles are supported by your device by running `rs-enumerate-devices`.

sam598, 2019-08-26:
Thanks @dorodnic for the quick reply.

To confirm I should not expect this capability to come to the D415 or D435?

dorodnic, 2019-08-26:
Yes.
Unfortunately it is not as simple as back-porting to existing firmware code, but relies on some hardware changes we are experimenting with.