realsenseai/librealsense #4726: How to Calibrate Depth Camera of D435I？

Garand0o0, 2019-08-26:
Hi,
I am experimenting with D435I.  I can use rtabmap to get 3d clouds. But in my point cloud, the ratio of the actual length of all objects to the measured length is a constant, such as 1.12. And this constant is related to the distance of the camera from the measured object. I found that the depth of d435i was inaccurate by `rs-distance`. How to Calibrate Depth Camera of D435I？Thanks.

MartyG-RealSense, 2019-08-26:
The depth sensing of 400 Series cameras can be calibrated with the Dynamic Calibrator software.

https://downloadcenter.intel.com/download/28517/Intel-RealSense-D400-Series-Calibration-Tools-and-API

If you are using Linux, installation instructions are on page 14 of the calibration user guide.

https://www.intel.com/content/www/us/en/support/articles/000026723/emerging-technologies/intel-realsense-technology.html

MartyG-RealSense, 2019-08-28:
Calibration can improve an image in some circumstances, and also correct inaccuracies in depth readings if the inaccuracies are related to the camera's calibration.  

If your project makes use of the D435i's IMU component, the IMU should be calibrated separately from the depth sensors.  If you do need to do this, the link below has an IMU calibration guide.

https://dev.intelrealsense.com/docs/depth-camera-d435i-imu-calibration

Garand0o0, 2019-08-28:
I want to improve the depth quality of D435I. The course you give is the calibration of dynamic parameters. I tried the calibration method you gave, but I didn't see any obvious effect. Are you sure that calibrating dynamic parameters will improve quality?

RealSenseCustomerSupport, 2019-09-20:
----------------------------------------------

Hi

Is this issue resolved? if not, how bad the the Depth accuracy? were you saying the error is 12%? Was it measured by our Depth Quality Tool (DQT)?
If not, please download and try the DQT tool from here: https://github.com/IntelRealSense/librealsense/releases/tag/v2.28.0

Thanks

closed 2019-09-28 by RealSenseCustomerSupport