realsenseai/librealsense #4728: Recording to .bag File without compression

pandurdo, 2019-08-26:
---------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435} | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version |   {Win 10}  |  
| SDK Version                            |  v2.25.0  | 
| Language                            |  {C++ }  | 

### Issue Description
I am trying to record to a .bag file without compression. Therefore I wrote a c++ sample, where I set some options ( framerate, laser power, etc.) and define a recording device as following:

%%%

    rs2::frameset frameset;
	rs2::context ctx;

	rs2::pipeline pipe(ctx);
	rs2::config cfg;

	rs2::device_list devlist = ctx.query_devices();
    rs2::device dev = devlist.front();
	// std::string devname = dev.get_info(RS2_CAMERA_INFO_NAME);
	// std::string devserial = dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER);

	auto depthsensor = dev.first<rs2::depth_sensor>();
	depthsensor.set_option(RS2_OPTION_DEPTH_UNITS, 0.0001);
	depthsensor.set_option(RS2_OPTION_LASER_POWER, atof(argv[2]));

	cfg.enable_stream(RS2_STREAM_COLOR, 848, 480, RS2_FORMAT_RGB8, atoi(argv[1]));
	cfg.enable_stream(RS2_STREAM_DEPTH, 848, 480, RS2_FORMAT_Z16, atoi(argv[1]));
	cfg.enable_device(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));

	std::time_t NOW = std::chrono::system_clock::to_time_t(std::chrono::system_clock::now());
	std::string s(20, '\0');
	std::strftime(&s[0], s.size(), "%d-%m-%Y_%H-%M-%S", std::localtime(&NOW));

    std::stringstream stream;
	stream << "Dev0_" << s.c_str() <<".bag";

	std::cout << "Saving frameset to " << stream.str() << std::endl;
	rs2::recorder rdev(stream.str().c_str(), dev, 0);
	pipe.start(cfg);
%%%

The penultimate line should define the device as recording device and disable compression by setting the last input to 0.

After executing it, my sample code creates the .bag file and starts taking frames but unfortunally doesn't save them to the file.

closed 2019-08-27 by pandurdo

codesian, 2019-08-28:
Hi @pandurdo!

How can you solve it?

Thanks

pandurdo, 2019-08-28:
I followed the instructions from here:

https://github.com/IntelRealSense/librealsense/issues/4499#issuecomment-514938979