realsenseai/librealsense #4736: Using Jetson TX2 with Realsense D435

MaryBD, 2019-08-27:
I need to use Realsense D435 with Jetson TX2. Is there any recommendation on how to do that. I tried flashing TX2 with Jetpack 4.2 but it wasn't compatible with librealsense release so I used Jetpack 3.3 and 3.2, but still had lots of issues with CUDA, ... and can't get the camera working. I appreciate any advices.

MaryBD, 2019-08-27:
Thanks for your comment. I got error when try to Install the libraries as:
E: Unable to locate package sudo apt-get install librealsense2-dkms

MaryBD, 2019-08-27:
Thanks for your advice. I finally installed Realsense D435 on TX2 following this: https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md
and then at "Prepare Linux Backend and the Dev. Environment" instead of what suggested for ubuntu 16 I followed ubuntu 14 and it works now. However the streaming is very slow and I see lots of errors while streaming.

dorodnic, 2019-08-27:
Sorry, skip the DKMS and just run: `sudo apt-get install librealsense2-utils`

dorodnic, 2019-08-27:
Please consider retrying `sudo apt-get install librealsense2-utils` since it will provide you with optimised SDK with CUDA. If you build from source, you need to make sure you build it with CUDA support.

MaryBD, 2019-08-28:
I faced some problem using CUDA, I am gonna give another try today. Thanks.

MaryBD, 2019-09-05:
I get it work: https://github.com/IntelRealSense/librealsense/issues/3617#issuecomment-528342960

closed 2019-09-16 by dorodnic

q8wwe, 2020-03-08:
bro  how you make it work ??

MaryBD, 2020-03-09:
@q8wwe I explained it here:
 https://github.com/IntelRealSense/librealsense/issues/3617#issuecomment-528342960