realsenseai/librealsense #4742: ctx.query_devices() always =0 on Android

calipos, 2019-08-28:
build with cmake.
CMAKE_TOOLCHAIN_FILE  is  android-ndk-r17c/build/cmake/android.toolchain.cmake.
FORCE_LIBUVC = TRUE.

i get the includes&librealsense2.a and BUILD my application SUCCESSFUL with the lib. 
but the ctx.query_devices() always =0 when the apk starts.
what the causal factors are? THANK YOU...

RealSenseCustomerSupport, 2019-09-17:
----------------------------------------------

@calipos What's your output of "adb shell lsusb"? And which SDK version are you using?

RealSenseCustomerSupport, 2019-09-24:
----------------------------------------------

@calipos Could you please update so that we can look into it further? Thanks!

calipos, 2019-09-29:
@RealSenseCustomerSupport   Thanks for your advice！I finally realised i lacked access permissions...

RealSenseCustomerSupport, 2019-10-12:
----------------------------------------------

@calipos Thanks for the update! Will close the issue accordingly.

closed 2019-10-12 by RealSenseCustomerSupport