realsenseai/librealsense #4743: Experiencing lower frame rate than requested in ROS2

limjizhengshawn, 2019-08-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | (05.10.13.00) | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.15.0-58-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.26.0                         | 
| Language                            |  C                          | 
| Segment			|  Others                        | 

### Issue Description
Hi,

I'm using:
librealsense commit master e5d51792182a71f47fa87e086fa46a9b778a06e8
ros2_intel_realsense commit master 31cdfb7
ROS 2.0 Dashing
Ubuntu 18.04

I have tried both compiling from source and installing through debian.

How to replicate:
Terminal 1: `ros2 run realsense_ros2_camera realsense_ros2_camera`
Terminal 2: `ros2 topic hz /camera/color/image_raw`

By default, the driver is supposed to produce images at 30fps.
However, the output at terminal 2 shows frequency lower than that. I've tried in the 3 different protocols:
FastRTPS - ~8-10Hz
OpenSplice - ~15-17Hz
RTI Connext - ~18-20Hz

Tested the method suggested by ros2 to test transmitting images:
`ros2 run image_tools cam2image -b -x 720 -y 1280 -f 30`
`ros2 run image_tools showimage`
`ros2 topic hz /image`

In FastRTPS I'm able to get up to 28 fps.

https://github.com/intel/ros2_intel_realsense/issues/67

RealSenseCustomerSupport, 2019-10-24:
----------------------------------------------

Hi @limjizhengshawn,

ROS 2 Dashing is not supported currently by Intel RealSense SDK 2.0. We are working on support for it, though no specific release date is known as of right now.

Thank you,
David

RealSenseCustomerSupport, 2019-10-30:
----------------------------------------------

@limjizhengshawn

Do you need any further clarifications regarding this topic?

closed 2019-11-05 by RealSenseCustomerSupport