realsenseai/librealsense #4744: Intel ResalSense D4xx Recovery

Samuel3Shin, 2019-08-28:
| | |
|---|---|
|**librealsense**|2.26.0 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense D4xx Recovery|
|**Product Line**|D400|
|**Asic Serial Number**|836313021804|

Please provide a description of the problem

Hi guys,

I'm currently using Ubuntu 16.
I tried firmware upgrade of my RealSense D435 and it failed.
After that, in realsense-viewer, my device is shown as a ResalSense D4xx Recovery

How can I fix it???? I want to make the device back to the normal version "RealSense D435".

MartyG-RealSense, 2019-08-28:
Intel recently introduced a new, easier firmware upgrade system that includes a tool called **rs-fw-update**.  This tool can do a recovery install on a camera with a failed firmware update.

https://dev.intelrealsense.com/docs/firmware-update-tool

To quote:

********

A camera/s might be in a recovery state, in such case listing the devices will output the following:

connected devices:
1) Name: Intel RealSense D4xx Recovery, serial number: unknown, ASIC serial number: 012345678901, firmware version: unknown, USB type: unknown

In such case we can use the recovery flag and run:

 **rs-fw-update -r -f Signed_Image_UVC_5_11_6_250.bin**

dorodnic, 2019-08-28:
Also, there is an option in the RealSense Viewer on the Recovery Device to Update to Recommended Firmware (under the "hamburger" menu)

Samuel3Shin, 2019-08-29:
Thank you guys!

Especially, @dorodnic your solution solved my problem with just one click.

closed 2019-08-29 by Samuel3Shin