realsenseai/librealsense #4747: Compiling error - Python Wrapper

danvica, 2019-08-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D415 } | 
| Firmware Version                   | (05.11.06.250) | 
| Operating System & Version |   {Win 10)  | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.0 }                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot }                          | 

### Issue Description

I've got a problem compiling the Python wrapper.

As in issues #1817 and  #4335, the error is:

CMake Error at CMakeLists.txt:16 (set_target_properties):

set_target_properties called with incorrect number of arguments.

CMake Error at CMakeLists.txt:57 (if):
if given arguments:

        "STREQUAL" "RS2_USE_V4L2_BACKEND"

Unknown arguments specified

I've already tried to execute cmake-gui from several path including "C:\Program Files (x86)\Intel RealSense SDK 2.0" and "C:\Program Files (x86)\Intel RealSense SDK 2.0\librealsense-master" but without any difference.

Any suggestion ? Many thanks.

grigala, 2019-08-28:
Have you tried to call CMake from top most source directory? I've had the same error when I was trying to compile it from `/wrappers/python/` directory, so try to CMake it from root folder.

danvica, 2019-08-28:
Thanks for the quick help.
Well, I've executed cmake-gui from the following paths:

C:\
C:\Program Files (x86)
C:\Program Files (x86)\Intel RealSense SDK 2.0
C:\Program Files (x86)\Intel RealSense SDK 2.0\librealsense-master
C:\Program Files (x86)\Intel RealSense SDK 2.0\librealsense-master\src
C:\Program Files (x86)\Intel RealSense SDK 2.0\librealsense-master\wrappers

Note: Every time deleting the cache.

Any suggestion ?

grigala, 2019-08-28:
I've just compiled it from the `master`. 
Must be a local issue, try to clone the source somewhere else than `C:\Program Files (x86)\Intel RealSense SDK 2.0` i.e. in your home directory or desktop and then in root folder:
```
mkdir build && cd build
cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true
```
P.S. Also make sure you have all the required dependencies installed. 
[image]

danvica, 2019-08-28:
Thanks, I'll try that.

BTW, I'm on windows 10.

grigala, 2019-08-28:
I know, me too :) 

Here if you didn't liked `WSL` shell 😄 

[image]

danvica, 2019-08-28:
It works!
Many thanks.

closed 2019-09-03 by dorodnic