realsenseai/librealsense #4748: D415 RGB Camera Specification

grigala, 2019-08-28:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D415 | 

### Since RGB sensor is a separate camera are intrinsics the same as depth intrinsics?
This might be a silly question but I'm interested in RGB camera specifications, like focal length, sensor size(36 x 24 mm?), nearest & farthest points and etc. I couldn't find any of this info neither in product specification(only mentions resolution and frame rate) nor in API. 

Thanks in advance.

MartyG-RealSense, 2019-08-28:
On the D415 model (the cased model and the caseless Depth Module Kit model) the RGB sensor is integrated on the camera circuit board.  It is not board-integrated on the other models in the 400 Series range.  On the cased camera models such as D435 that have RGB but not a D415 board (the D435 models use the 'D430' board), the RGB sensor is attached separately via cable inside the camera casing.

The details of the color sensor (the OmniVision OV2740) are listed on page 36 of the current edition of the data sheet document for the 400 Series cameras, including the focal length.  On the D415, the color sensor and the IR sensor use the same OV2740 sensor.  On the D435 models, a wider OV9282 sensor is used for the IR.

https://www.intel.com/content/www/us/en/support/articles/000026827/emerging-technologies/intel-realsense-technology.html

[image]

grigala, 2019-08-28:
Oh my bad, I missed that page apparently. 

Thanks @MartyG-RealSense

closed 2019-08-28 by grigala