realsenseai/librealsense #4750: Reproduction of hanging D400 camera issue

pwalczykk, 2019-08-28:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.11.01.00 | 
| Operating System & Version | Ubuntu 16.04 | 
| Kernel Version (Linux Only)    | 4.4.0-131-generic | 
| Platform                                 | PC |
| SDK Version                            | 2.1.0 | 
| Language                            | C++(ROS driver) | 
| Segment			| Robot | 

Hi,

I have prepared an experiment that allows to reproduce the issue of hanging camera mentioned in [this thread](https://github.com/IntelRealSense/librealsense/issues/4552). 

The experiment is launching RealSense camera ROS nodes in Docker container on the host computer and terminates it after a short time. If some camera is missing, it applies XHCI USB driver restart or shutdown and boot delayed with wakealarm of the PC to restart cameras. Everything happens in the loop controlled by the client computer. Source code with more detailed description of the experiment is available in [this repo](https://github.com/NoMagicAi/reproduce_d400_camera_issue)

The experiment gathers logs from roslaunch and rs-fw-logger, which might be useful for debugging purpose. A sample of logs from 64-hour experiment, which contains 45 examples of hanging camera issue is available under [this link](https://drive.google.com/file/d/1oQOymr_f1ylLZ8w_Q1kWSmGma9xG92Q9/view?usp=sharing) 

I hope this will bring us a bit closer to solve the issue of hanging D400 cameras.

MartyG-RealSense, 2019-08-28:
Thank you very much for your carefully prepared test!  Well-structured test result data is a useful resource for increasing technical understanding of why certain problems might occur.  The commitment of yourself and other community members to improving the experience of RealSense users is really appreciated.  :)

RealSenseCustomerSupport, 2019-11-01:
----------------------------------------------

We will continue tracking this ticket through Zendesk portal.
Closing this one unless otherwise.
Thank you.

closed 2019-11-09 by RealSenseCustomerSupport