realsenseai/librealsense #4753: Android GLRsSurfaceView single frame

Murded, 2019-08-29:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.11.06.200 | 
| Operating System & Version | Android 9  | 
| Platform                                 |Phone  |
| SDK Version                            |  implementation 'com.intel.realsense:librealsense:2.+@aar                         | 
| Language                            | Java (Android)  | 

### Issue Description
I've been using one of the examples and the following code is used to display the different streams:

```
 try
            {
                try(FrameSet frames = mPipeline.waitForFrames(1000))
                {
                    try(FrameSet processed = frames.applyFilter(mColorizer))
                    {
                        mGLSurfaceView.upload(processed);
                        mHandler.post(mStreaming);
                    }
                }
            }
            catch (Exception e) {
                Log.e(TAG, "streaming, error: " + e.getMessage());
            }
```

Which works fine, the depth and colour streams are both displayed, however I wanted to only display the colour stream so I changed the method so this instead (Just passing the colour frame instead of the frameset)

```
 try
            {
                try(FrameSet frames = mPipeline.waitForFrames(1000))
                {
                    try(FrameSet processed = frames.applyFilter(mColorizer))
                    {
                        mGLSurfaceView.upload(frames.first(StreamType.COLOR));
                        mHandler.post(mStreaming);
                    }
                }
            }
            catch (Exception e) {
                Log.e(TAG, "streaming, error: " + e.getMessage());
            }
```

However after only a few frames the stream freezes and I see this message in the Logcat 
```
Failed to send DDMS packet REAQ to debugger (-1 of 20): Broken pipe
```

Any suggestions? I'm asssuming that perhaps the frames aren't being released or something? I've tried manually calling .close on the processed frameset at the bottom but that makes no difference and android actually says it redundant because the processed frameset is inside the try block anyway

Murded, 2019-08-29:
After looking at the Debug logs I can see this is displayed over and over:
```
2019-08-29 12:25:04.510 29169-29249/com.gpcsl.AndroidRealsense I/librs: Dropped frame. alloc_frame(...) returned nullptr
2019-08-29 12:25:04.569 29169-29249/com.gpcsl.AndroidRealsense I/chatty: uid=10166(com.gpcsl.AndroidRealsense) identical 1 line
```

Murded, 2019-09-02:
Anyone else seen this issue?

Murded, 2019-09-05:
@matkatz Could you verify this is an issue please and I'm not just doing something wrong?

RealSenseCustomerSupport, 2019-10-17:
----------------------------------------------

Hi @Murded,

Several questions:
1) Did you refer to below capture example to build your own app?
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android/examples/capture
2) which librealsense version do you use?
3) Per your info, android 9 SDK is used, right?

Murded, 2019-10-17:
> Hi @Murded,
> 
> Several questions:
> 
> 1. Did you refer to below capture example to build your own app?
>    https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android/examples/capture
> 2. which librealsense version do you use?
> 3. Per your info, android 9 SDK is used, right?

Hi,

I actually didn't refer to that example but a different one (I can't remember which) however since I needed extended functionality I've found a different way to acheive what i wanted.  

In regards to the issue, the example you posted is what I was trying to acheive so It is likely that I was doing something (Most likely not disposing of something).

I don't recall the realsense veresion, I just added it as a dependency to my build.gradle at the time this ticket was created.

That's correct, Operating System version Android 9 (API level 28).

I no longer require anything related to this issue so please feel free to close.

closed 2019-10-18 by RealSenseCustomerSupport