realsenseai/librealsense #4755: Firmware image (version 3.26.1.0) for SR300

noell1, 2019-08-29:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { SR300 } | 
| Firmware Version                   | (3.26.1.0) | 
| Operating System & Version |   {Linux (Ubuntu 16) | 
| Kernel Version (Linux Only)    |  (e.g. 4.15.0)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.26.0 }                          | 
| Language                            |  {C }                          | 
| Segment			|  {others }                          | 

I would like to update firmware  on all SR300 cameras to latest version (3.26.1.0). Using realsense-viewer is quite time consuming. Is it possible to perform firmware update of SR300 with intel-realsense-dfu?
 If yes, where can I find (download) latest firmware image? On https://downloadcenter.intel.com/  only firmwares for D400  are available.

MartyG-RealSense, 2019-08-29:
The recommended firmware version for the SR300 is now **3.10.10.0**

I don't know of an ability to update the SR300 with the current DFU tool.  The older Windows-based SDK from 2016 had a text-based DFU tool for the SR300, though this may not be useful to you if you can only update via Linux.

https://forums.intel.com/s/question/0D70P000006973mSAA

I do not have a source for the firmware file though.  If it is updatable through the RealSense Viewer, I would speculate that Intel might have bundled it into the RealSense Viewer's package now.

matkatz, 2019-09-02:
[rs-fw-update tool](https://github.com/IntelRealSense/librealsense/tree/master/tools/fw-update) is a command line tool that performs DFU process for both D400/SR300 cameras.
This tool is part of the [installer](https://github.com/IntelRealSense/librealsense/releases/download/v2.27.0/Intel.RealSense.SDK-WIN7.exe).

The current recommended FW for SR300 (same as the one that is embedded in the viewer executable ) is [3.26.1.0](http://realsense-hw-public.s3-eu-west-1.amazonaws.com/Releases/SR300/FW/SR3XX_FW_Image-3.26.1.0.bin).

MartyG-RealSense, 2019-09-02:
The situation with the SR300 firmware is kinda confusing.  The release notes for SDK 2.25.0 recommended that firmware 3.26.10 be used, but the Release page of the same version suggests firmware 3.10.10.0 for the SR300, as does every version up til the latest SDK version 2.27.0.

[image]

I note the **+** beside the version name, suggesting that SR300 firmware version 3.10.10.0 **and newer** is recommended.  Without a link to a firmware release page like the 400 Series firmwares have so the firmwares can be listed by release date, it's impossible for users to know this though.

Also, a note on the Release page explaining that the firmwares are bundled into the Viewer would be highly useful.

noell1, 2019-09-03:
Thank you both for clarification and latest FW for SR300.

closed 2019-09-03 by noell1

Chenxin9607, 2021-05-02:
> [rs-fw-update tool](https://github.com/IntelRealSense/librealsense/tree/master/tools/fw-update) is a command line tool that performs DFU process for both D400/SR300 cameras.
> This tool is part of the [...]

Thanks for your 3.26.1.0 firmware, but where can i get 3.10.10.0 firmware.

MartyG-RealSense, 2021-05-02:
Hi @Chenxin9607  SR300 firmware updates are not available as individual downloads like the 400 Series firmwares are, so you cannot update to a firmware other than the **3.26.1.0** one installed through the RealSense Viewer.

Chenxin9607, 2021-05-02:
> Hi @Chenxin9607 SR300 firmware updates are not available as individual downloads like the 400 Series firmwares are, so you cannot update to a firmware other than the **3.26.1.0** one installed through the RealSense Viewer.

thanks :)