realsenseai/librealsense #4756: RealSense SDK & D435

coldcoda, 2019-08-29:
Hello,
Does the Intel RealSense SDK (not SDK2) support the depth camera D435? And can I use the SDK to complete the Face Tracking and Recognition.

Thanks a lot !!!

MartyG-RealSense, 2019-08-29:
The 2016 SDK for Windows (2016 R3) does not support the 400 Series cameras such as D415 and D435, unfortunately.  It supports F200 and SR300 cameras.

coldcoda, 2019-08-29:
I wish to get face landmarks with D435 also use unity to complete it, Is there any functions provided by Realsense SDK2 can  help me? I also wish to get Three-dimensional information of human face. 

Finally, what I want to accomplish is that the change of facial expression drives the change of virtual face model. So,Is there any functions in Realsense SDK2 can  help me?

MartyG-RealSense, 2019-08-29:
Nuitrack SDK supports face landmarks and can also be tightly integrated with Unity.  Below is a link to a Nuitrack face tracking tutorial for Unity that changes the expression on a 2D face sprite.

https://download.3divi.com/Nuitrack/doc/UnityFaceTracking_page.html

There is also face detection software in Unity based on the OpenCV vision software.  It is not inexpensive though, and its most advanced features require an additional costly OpenCV program.

https://assetstore.unity.com/packages/tools/integration/dlib-facelandmark-detector-64314

Overall, the RealSense SR305 camera is a better choice than a 400 Series camera for face detection and tracking applications and for controlling a virtual 3D avatar in Unity.  You should certainly be able to achieve some form of face landmark tracking using one of the above methods though.

RealSenseCustomerSupport, 2019-09-24:
----------------------------------------------

Closing this ticket unless there is any further question. Thank you.

closed 2019-09-30 by RealSenseCustomerSupport