realsenseai/librealsense #4760: Will the Box Dimensioner Example work for SR300? (python)

skylermaxwell42, 2019-08-29:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Firmware Version                   | 3.26.0 | 
| Operating System & Version |  Windows 10 |                                        | 
| Platform                                 | PC  |
| SDK Version                           |                  | 
| Language                               | Python  | 

### Issue Description
I have followed the box dimensioner example using the instructions in the README. I made a few adjustments to the example to get things "sort-of" working which are listed below.

(1) box_dimensioner_multicam_demo.py - lines (108-109)

Commented out the section where it loads presets for the RS400 advanced mode which does not apply to the SR300 camera.

```python
# Load the JSON settings file in order to enable High Accuracy preset for the realsense
#device_manager.load_settings_json("./HighResHighAccuracyPreset.json")
```

(2) realsense_device_manager.py - lines (169-181)

Refactored the section where IR is enable and disabled to account for differences between the SR300 and RS400.

```python
    def enable_emitter(self, enable_ir_emitter=True):
        """
        Enable/Disable the emitter of the intel realsense device
        """
        for (device_serial, device) in self._enabled_devices.items():
            # Get the active profile and enable the emitter for all the connected devices
            sensor = device.pipeline_profile.get_device().first_depth_sensor()
            #sensor.set_option(rs.option.emitter_enabled, 1 if enable_ir_emitter else 0)
            if enable_ir_emitter:
                sensor.set_option(rs.option.laser_power, 16)
            else:
                sensor.set_option(rs.option.laser_power, 0)
```
(3) realsense_device_manager.py - lines (155-157)

Same reason as (2)

```python
sensor = pipeline_profile.get_device().first_depth_sensor()
sensor.set_option(rs.option.laser_power, 16 if enable_ir_emitter else 0)
self._enabled_devices[device_serial] = (Device(pipeline, pipeline_profile))
```
------------------
I am using a 2 camera set up and the example is now able to calibrate the devices to the world coordinates using the checkerboard but it is not able to detect any box placed onto the checkerboard. 

What else can I do to try and get this example to work? Will this example even work with the SR300 camera? 

Thanks for any input.

MartyG-RealSense, 2019-08-29:
I do not know personally whether there is a reason why the script would not work with SR300.   However ...

Are you using the 6x9 calibration chessboard size specified in the instructions as your calibration object, please?   If you use a different size then you need to adjust the chessboard parameters in the box dimensioner script.

Also, the box that is being measured should be smaller than the size of the calibration object.

skylermaxwell42, 2019-08-29:
@MartyG-RealSense 

Hi Marty,

Thanks for your quick response.

One reason it will not work is because when you run 
```python
device_manager.load_settings_json("./HighResHighAccuracyPreset.json")
```
it fails because the SR300 does not accept these options. There are other places in the code where it must be modified as well because it fails on certain SDK calls that only apply to the RS400 cameras

I am using the provided checkerboard in the project and I have the cameras less than 1 meter from the checkerboard. The box also fits within the checkerboard. 

Since the example has both cameras running, is it possible that the coded light stream is interfering with each other camera?

MartyG-RealSense, 2019-08-29:
SR300 cameras can interfere with each other when placed in proximity to one another, yes.

I'm not sure that it needs to have a json loaded.  I'm not a Python expert but with SDK 2.0 in general, if you do not specify a configuration with a config / cfg command then the SDK uses that camera's default configuration.

dorodnic, 2019-08-29:
Unfortunately, coded-light cameras (SR300) do interfere with each other in quite severe way when sharing field-of-view.

MartyG-RealSense, 2019-08-29:
There is Microsoft's 2012 'shake n sense' method of breaking up interference by vibrating a structured light camera with motors (Kinect 1 used coded light, whereas Kinect 2 was time of flight).   Though in honesty, I haven't heard of anyone who ever got the technique to work with a RealSense F200 or SR300 camera!

 https://www.microsoft.com/en-us/research/publication/shakensense-reducing-interference-for-overlapping-structured-light-depth-cameras/

skylermaxwell42, 2019-09-03:
@dorodnic Thank you for the confirmation about the SR300 interference. 

@MartyG-RealSense Thanks for the suggestion but that approach is not feasible for me.

closed 2019-09-03 by skylermaxwell42