realsenseai/librealsense #4761: use rs-convert to get csv from bag, failed

JingqiuZuo, 2019-08-29:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I got a .bag file from D435. And I want to use the rs-convert tool to change it into a csv file.
I enter the command: 
`./rs-convert -v ~/librealsense/build/tools/convert/test.csv -i ~/librealsense/build/tools/convert/subset.bag`
But get the following error:
`RealSense error calling rs2_pipeline_start_with_config(pipe:0x8af160, config:0x8af1a0):
    Failed to resolve request. Request to enable_device_from_file("/home/turtle/librealsense/build/tools/convert/") was invalid, Reason: Failed to create ros reader: Error reading version line`
Is it because of I use a wrong command?
Or is the usage wrong?
How to use the rs-convert tool in Ubuntu16?

Thanks in advance!

RealSenseCustomerSupport, 2019-09-24:
----------------------------------------------

Hi @JingqiuZuo,

Please try to convert this bag file to CSV file(s) in your environment,using the rs-convert tool and let us know the results:
http://realsense-hw-public.s3.amazonaws.com/rs-tests/TestData/stairs.bag

Thank you!

RealSenseCustomerSupport, 2019-10-03:
----------------------------------------------

@JingqiuZuo -

Do you need any further clarifications regarding this topic?

Thank you!

closed 2019-10-22 by RealSenseCustomerSupport