realsenseai/librealsense #4762: Bag unindexed error when recording

realsens01, 2019-08-29:
Creating a new issue because [2835](https://github.com/IntelRealSense/librealsense/issues/2835) was closed without resolution.

Notes:
* See [3044](https://github.com/IntelRealSense/librealsense/issues/3044#issuecomment-464514016) for my system specification.
* Dorodnic referred to [2835](https://github.com/IntelRealSense/librealsense/issues/2835) as a known and unresolved issue in [3029](https://github.com/IntelRealSense/librealsense/issues/3029#issuecomment-453114359).
* My solutions so far have been to destroy all Realsense objects when I stop recording -- which isn't a great solution, but helps -- and to use ROS to reindex the bags (time consuming) when that doesn't work.
* I suspect that the reason this is still sometimes an issue in my application -- even after following the advice to explicitly destroy all Realsense objects (a step that I'm not thrilled with) -- is because users are ending the application process before the bag is indexed. I believe this is because bag indexing takes some time, and that the time scales with the size of the bag. I need to confirm this, but I think I'm only seeing the unindexed error for larger bags. Our recordings can be 20 minutes or longer. When the user stops recording, the Realsense python objects are destroyed, so I can't check on their status. What I think is happening is that the bags are still being processed when the application closes, but I have no way of knowing this because the objects are destroyed (necessary to start indexing). I might be totally wrong about this, but that's my current theory.

RealSenseCustomerSupport, 2019-09-24:
----------------------------------------------

Hi @realsense01,

To clarify:
1. For your solution, you are doing "pipe.stop()" if you stop recording in order to resolve this issue, right?
2. What's the librealsense version you currently use?

realsens01, 2019-09-25:
1. I don't understand your question. I _expect_ the file to be written when I call ``pipeline.stop()``, but I continue to get unindexed errors when I try to read the bag. Previously, I thought that the solution was to destroy all objects related to the pipeline object, but now I am not so sure.
2. I do not have direct access to that information at the moment, but the last version that I am aware of using is 2.20. It's unlikely that I've upgraded since then, since I am trying to keep the system stable.

My current theory is that a separate thread or process is spawned to write the bag file when I call ``pipeline.stop()``, and that I am somehow preventing it from writing the bag index when I exit the script early. However, I have no way of knowing how much progress it has made in writing the bag index, so I cannot warn the user to not terminate the program. Do you know of a way to get this information? Is anyone there familiar enough with the bag writing process to evaluate this theory?

RealSenseCustomerSupport, 2019-11-05:
----------------------------------------------

hi @realsens01,

I hope the community can also help this bag unindexed issue withe better error handling especially because of unexpected record stop. For example, ctrl + c to break the app. As you mentioned, use the rosbag command to reindex the file as a workaround first.

RealSenseCustomerSupport, 2020-01-22:
----------------------------------------------

Hi @realsens01

There hasn't been any posts on this thread.

realsens01, 2020-01-23:
That's true.

jagatfx, 2020-02-06:
I can ensure the comprehensiveness of this but what worked for me was to ensure in addition to calling `pipe->stop()` be sure to set the `pipe = nullptr`.

realsens01, 2020-02-06:
@DavidCadtec No solution has been offered for the cause of the unindexed error. I've put measures in place to introduce delays that give the bag time to write, but I still see the error fairly regularly. As I noted in [an earlier post](https://github.com/IntelRealSense/librealsense/issues/3044#issuecomment-464514016), the rosbag utility can be used to fix bags after they've been recorded. This is how I currently handle it, but it is time consuming (and it is not a solution to the problem with Intel's implementation).

@jagatfx This does not work in my scenario, as explained above.

DavidCadtec, 2020-02-06:
Thanks for your input @realsens01 
Where do i get this rosbag utility ?

DavidCadtec, 2020-02-06:
@realsens01
I added the calls to destroy pipeline and config objects and attributed None to them and everything worked perfectly with the recording !
As you said... not an awesome solution .. but it's a way to at least get the bag file properly created (not with unindexed issue ...).
Thank you !

closed 2020-02-07 by RealSenseCustomerSupport

realsens01, 2020-02-07:
@RealSenseCustomerSupport: @DavidCadtec believed his issue was solved. I do not. I opened this issue. Why did you close it without a solution?

realsens01, 2020-02-07:
@DavidCadtec: There is a rosbag package on the latest version of Ubuntu, iirc. However, I think you might have to install ROS for 16.04.

I would be careful about believing you've solved the problem with the strategy I mentioned. Have you tried allowing a bag to record for an extended period, and then reproducing the closure? For me, the problem seemed solved until there was long enough of a delay for the bag to take time while writing to disk.

DavidCadtec, 2020-02-10:
@realsens01 
I did not mention in my post that "i've solved the problem" ... i just stated (and repeat here...): "it's a way to at least get the bag file properly created ..."
Just by calling the pipeline.stop should already be enough to generate the bag file properly ... So (i agree with you...) the issue remains open ...
But now, at least, i can get bag files with the videos recorded ... 
My sequences are short here ...  so worked for me ... but per your comment i understand that there are still issues when recording long videos ... is this so ?

realsens01, 2020-02-10:
@DavidCadtec : Sorry. It's just my experience that the "support" here closes issues if there's even a hint that something has been solved. In the 1-2 years I've been using the product, I'm not sure they've ever actually resolved an issue. It's really frustrating.

>  i just stated (and repeat here...): "it's a way to at least get the bag file properly created ..."

Sorry. Thanks for clarifying.

>  i understand that there are still issues when recording long videos ... is this so ?

See [my earlier post](https://github.com/IntelRealSense/librealsense/issues/4762#issue-487120855). Setting the objects to None seem to trigger the file writing / indexing process. However, that still takes some time. For long recordings / big files, it takes longer. In the context of the project I'm working on, users are collecting data with the Realsense cameras for 20 minutes or more, so the time to write is not insignificant. What I believe is happening is that the users stop the recordings and then move on, sometimes closing the application while the writing process is still happening, which results in "unindexed" errors. Once you've set the objects to None (to trigger the write), there doesn't seem to be any way to check on the status of the write, so the only way to avoid this error is to impose a fixed delay, and hope it's enough.

There are likely other solutions, but at this point I can't justify spending more time debugging Realsense-related code. It's only supposed to be a small part of this project, but has instead dominated the development process. I'd prefer they just fix it.

DavidCadtec, 2020-02-22:
@realsens01 : again thanks for you clarifications and details related to this issue.  Yes, i fully agree with you that there should be a solution to avoid this issue happening in recording .bag files.

For example, look into my condition here: i need to move ahead with an academic research and i do need a way to record materials (RGBD sequences with people walking ...) so that i can use in my algorithms ... i tried to find some .bag files already recorded of pedestrian walking around but had no success ... so i need to record myself... 
Again thank you for your comments and suggestions.