realsenseai/librealsense #4763: apt-get install librealsense2-dkms produces error and unknown result on system with multiple kernels available

eric-schleicher, 2019-08-29:
|                 Item  |  Value |          
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  Series | 
| Firmware Version                   | N/A | 
| Operating System & Version |   Ubuntu 18.04| 
| Kernel Version (Linux Only)    |   4.18.0-15-generic                                       | 
| Platform                                 | PC  |
| SDK Version                            | 2.x (latest as of 2019-08-29) | 
| Language                            |  N/A| 
| Segment			|  Robots!                          | 

### Issue Description
On host system with more than one kernel available (including one not supported by DKMS packages)  installation script fails when trying to build modules for unsuppored kernel.  

Leaves user not knowing whether process was successful (for the kernel versions that **are** supported)

It's possible this is the expected behavior, but perhaps more insightful messaging for theuser to understand that the process was partially successful (?)

#### Console output
```
rosuser@panda-cap:~$ sudo apt-get install librealsense2-dkms
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libamd2 libbtf1 libcamd2 libccolamd2 libcholmod3 libcxsparse3
  libfreenect-dev libfreenect0.5 libgraphblas1 libklu1 libldl2 libllvm7
  libmetis5 librbio2 libspqr2 libsuitesparse-dev libumfpack5
  ros-melodic-costmap-2d ros-melodic-find-object-2d ros-melodic-libg2o
  ros-melodic-move-base-msgs ros-melodic-octomap ros-melodic-octomap-msgs
  ros-melodic-voxel-grid
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  dkms librealsense2-udev-rules
Suggested packages:
  menu
The following NEW packages will be installed:
  dkms librealsense2-dkms librealsense2-udev-rules
0 upgraded, 3 newly installed, 0 to remove and 23 not upgraded.
Need to get 4,196 kB of archives.
After this operation, 22.4 MB of additional disk space will be used.
Do you want to continue? [Y/n] Y                                       
Get:1 http://us.archive.ubuntu.com/ubuntu bionic-updates/main amd64 dkms all 2.3-3ubuntu9.5 [68.1 kB]
Get:2 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 librealsense2-dkms all 1.3.6-0ubuntu0 [4,122 kB]
Get:3 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 librealsense2-udev-rules amd64 2.26.0-0~realsense0.1436 [6,600 B]
Fetched 4,196 kB in 2s (1,791 kB/s)                 
Selecting previously unselected package dkms.
(Reading database ... 238438 files and directories currently installed.)
Preparing to unpack .../dkms_2.3-3ubuntu9.5_all.deb ...
Unpacking dkms (2.3-3ubuntu9.5) ...
Selecting previously unselected package librealsense2-dkms.
Preparing to unpack .../librealsense2-dkms_1.3.6-0ubuntu0_all.deb ...
Unpacking librealsense2-dkms (1.3.6-0ubuntu0) ...
Selecting previously unselected package librealsense2-udev-rules:amd64.
Preparing to unpack .../librealsense2-udev-rules_2.26.0-0~realsense0.1436_amd64.deb ...
Unpacking librealsense2-udev-rules:amd64 (2.26.0-0~realsense0.1436) ...
Setting up librealsense2-udev-rules:amd64 (2.26.0-0~realsense0.1436) ...
Setting up dkms (2.3-3ubuntu9.5) ...
Processing triggers for man-db (2.8.3-2ubuntu0.1) ...
Setting up librealsense2-dkms (1.3.6-0ubuntu0) ...
Loading new librealsense2-dkms-1.3.6 DKMS files...
Building for 4.18.0-15-generic 5.0.0-25-generic
Building initial module for 4.18.0-15-generic
Can't load /var/lib/shim-signed/mok/.rnd into RNG
140304523903424:error:2406F079:random number generator:RAND_load_file:Cannot open file:../crypto/rand/randfile.c:88:Filename=/var/lib/shim-signed/mok/.rnd
Generating a RSA private key
.................................................................+++++
..............................................................+++++
writing new private key to '/var/lib/shim-signed/mok/MOK.priv'
-----
This system doesn't support Secure Boot
Secure Boot not enabled on this system.
Done.

uvcvideo:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.18.0-15-generic/updates/dkms/

videodev.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.18.0-15-generic/updates/dkms/

hid-sensor-accel-3d.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.18.0-15-generic/updates/dkms/

hid-sensor-gyro-3d.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.18.0-15-generic/updates/dkms/

depmod...

DKMS: install completed.
Building initial module for 5.0.0-25-generic
Error! Bad return status for module build on kernel: 5.0.0-25-generic (x86_64)
Consult /var/lib/dkms/librealsense2-dkms/1.3.6/build/make.log for more information.

Loading the modified modules into kernel... complete
Current status:
librealsense2-dkms, 1.3.6, 4.18.0-15-generic, x86_64: installed
mod: videodev 			 version: realsense2-dkms
mod: uvcvideo 			 version: 1.1.2.realsense-1.3.6
mod: hid_sensor_gyro_3d 	 version: realsense2-dkms
mod: hid_sensor_accel_3d 	 version: realsense2-dkms
rosuser@panda-cap:~$ uname -a
Linux panda-cap 4.18.0-15-generic #16~18.04.1-Ubuntu SMP Thu Feb 7 14:06:04 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

```

kroq-gar78, 2019-08-30:
Not a dev on this project, but uploading the relevant bits of `/var/lib/dkms/librealsense2-dkms/1.3.6/build/make.log` would likely help.

ev-mp, 2019-09-02:
@eric-schleicher , during installation DKMS will automatically try to install the package for the currently running kernel (e.g. 4.8.0-15) as well as the most advanced version found in grub (5.0.0-25).
I'm not sure if it is feasible to change the default DKMS settings and avoid installation for non-supported kernels in the first place, but it can be explored. 
However it is a corner case which produces visual `artefact` but doesn't affect functionality, thus a low priority.

regarding the installation status - at the end of DKMS deployment the installation status will be flushed to console, as appears at the end of the attached log:
>Current status:
librealsense2-dkms, 1.3.6, 4.18.0-15-generic, x86_64: installed
mod: videodev 			 version: realsense2-dkms
mod: uvcvideo 			 version: 1.1.2.realsense-1.3.6
mod: hid_sensor_gyro_3d 	 version: realsense2-dkms
mod: hid_sensor_accel_3d 	 version: realsense2-dkms

It shows all DKSM packages/kernels installed as well as the version of the kernel modules "touched" by `librelasense2-dkms`. The user can immediately assess whether the installation was successful.

eric-schleicher, 2019-09-05:
thanks.  closing.

closed 2019-09-05 by eric-schleicher