realsenseai/librealsense #4764: How to resize existen bag file?

sailfish009, 2019-08-30:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version |   Win 10 64bit  | 
| Kernel Version (Linux Only)    |                                          | 
| Platform                                 | PC  |
| SDK Version                            |  2.26.0                   | 
| Language                            | C++                       | 
| Segment			|  VR/AR/others                       | 

### Issue Description

i am using [recorder](https://github.com/IntelRealSense/librealsense/tree/master/tools/recorder) from sdk, changed recording time 10s to 1s.
after recording, bag file size is about 30 ~ 40 Mbytes(auto exposure), 40 ~ 60 Mbytes(manual). (color 1920x1080, depth 1280x720, 15fps).
I think there are 10 to 15 frames of image files in the bag file. To check if the bag file was saved successfully, we tried to extract the jpg image from the bag file to see if the jpg file was saved normally. In most cases I have verified that the bag file is saved normally. The only remaining problem is
I want to reduce the size to 4-5 Mbytes by leaving only 1 frame in this bag file.

MartyG-RealSense, 2019-08-30:
ROS has a toolkit called **bag_tools ** that features a program called **cut.py**. It can input a bag file, crop it to the specified start point and 'duration' end-point, and write the result as an output bag.

http://wiki.ros.org/bag_tools#cut.py

sailfish009, 2019-08-31:
@MartyG-RealSense , thanks for the info!

closed 2019-08-31 by sailfish009