realsenseai/librealsense #4765: pyrealsense2 & python3.7

coldcoda, 2019-08-30:
Hello,
Does pyrealsene2 support python3.7 ?
I have installed python 3.7 ,But I can't install pyrealsense2. Do I have to install a python3.6 ?

MartyG-RealSense, 2019-08-30:
Python 3.7 support was added in SDK version 2.23.0 in June 2019.

https://github.com/IntelRealSense/librealsense/pull/4012

Support for Pypi installation of 3.7 seems to be something that is still in the works.

https://github.com/IntelRealSense/librealsense/issues/4515#issuecomment-521235404

lramati, 2019-09-01:
Your options are either to [compile from source](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#building-from-source) or to downgrade to Python 3.6 and install via PyPi. Keep in mind that PyPi installation is not currently supported on any version of Python for Mac.