realsenseai/librealsense #4766: bag files with missing frames hangs when playback is non-realtime.

Dealerpriest, 2019-08-30:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version |  Win 10  | 
| Platform                                 | PC |
| SDK Version                            | build 2.26.0                          | 
| Language                            |  C#/unity                          | 
| Segment			| VR   | 

When setting playback to **non-realtime**, paying back files with missing frames sometimes hangs.
To be clear. If a particular recording with missing frames hangs at a specific time (and frame number) it will hang every time. But some other missing frames will not cause the playback to hang. My impression is that if there's a lot of missed frames, like one second, it will hang, and if it's just a small "glitch" it will play through it.

**Note:** The reason for missing frames is probably data congestion during recording (I'm recording from 4 cameras simultaneously).

Dealerpriest, 2019-08-30:
One possibility I looked into was to use the rosbag inspector. But there isn't any compiled binary version included in the SDK install. Why?
I tried to compile the SDK myself but wasn't very successful. Anyone got any advice on this? Or perhaps a binary version of the rosbag inspector compiled for winx64?

Dealerpriest, 2019-09-02:
I have now tried this in both C++ and in Unity/C#. The behaviour is the same for both. So my conclusion is that there is some problem in the native library.

RealSenseCustomerSupport, 2019-09-26:
----------------------------------------------

Looks like it's related to https://github.com/IntelRealSense/librealsense/pull/4758
Please kindly take a look. Thank you.

closed 2019-10-05 by RealSenseCustomerSupport