realsenseai/librealsense #4767: 435i see as 430i

kinetixx, 2019-08-30:
Hello 

I have 6 cameras, RealSense 435i, all have the newest firmware -> D400_Series_Production_FW_5_11_6_250. 4 cameras the system see as 435i and two as 430i.
Serial numbers of these: 843112073212, 843112070069. RealSense Viewer see them as product id : 0B4B, but 435i see as 0B4A, but on the cameras it is written 435i. Is it possible? 

Thank you.

MartyG-RealSense, 2019-08-30:
430i isn't a camera model that is available for the public to buy yet, though it is supported by the SDK and so is probably a model of D435i without a casing (like the caseless D430 camera board) that Intel are doing internal testing on.

Typically, when a D435 mis-identifies as a D430, it is because the RGB sensor is not being detected.  I believe it may be corrected by downgrading the firmware to a slightly older version and then re-upgrading the firmware to the new version.

dorodnic, 2019-08-31:
It is possible if the cable connecting D400 ASIC board to the RGB module gets loose.
If you see this issue consistently I'd recommend contacting support for replacement.

ev-mp, 2019-09-18:
@kinetixx , what is the current status - did you turn in the unit for replacement ?

kinetixx, 2019-09-18:
We finally have 3 cameras as 430i. We decided to put them into projects where we just need depth. We bought next two 435i but they seem to work. We tried everything and it seems like it was hardware problem.

closed 2019-09-18 by kinetixx