realsenseai/librealsense #4768: RuntimeError: command transfer failed to execute bulk transfer, error: RS2_USB_STATUS_ACCESS

paolorota, 2019-08-30:
I have installed pyrealsense2 on my ubuntu 18.04 via pip3.

Once I run my code it says the error in the subject:

`RuntimeError: command transfer failed to execute bulk transfer, error: RS2_USB_STATUS_ACCESS` when I start the pipeline.

What does that mean? 

My camera is a Creative blasterX senz3D.

Thanks,
P.

MartyG-RealSense, 2019-08-30:
I just wanted to add for others who may have knowledge of this subject that the Creative BlasterX Senz3D is a RealSense SR300 camera manufactured by Creative (who also manufactured the official SR300 Developer Kit camera on behalf of Intel).  It looks and works the same as an official SR300.

dorodnic, 2019-08-31:
You can install realsense2-utils package or manually follow the installation step describing udev-rules update

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

dorodnic, 2019-08-31:
(Basically by default on Linux you don’t have permissions to work with the camera)

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

closed 2019-09-10 by dorodnic