realsenseai/librealsense #4769: Announcement: Highlighting the 'FRAMOS Depth Camera D435e' RealSense industrial camera

MartyG-RealSense, 2019-08-30:
Hi everyone,

It has come to my notice that Intel Authorized Distributor company FRAMOS, a long-time supporter of RealSense technology, have developed an industrial variant of the D435 named the 'FRAMOS Depth Camera D435e'.

There is established precedent for this - in the previous generation of cameras before the 400 Series, companies including Creative and Razer developed own-brand versions of RealSense cameras, such as the SR300-compatibles Creative BlasterX Senz3D and the Razer Stargazer.

Special features of the FRAMOS D435e aimed at industrial applications include water and dust resistant housing and M12 ethernet and M8 power connectors.

The camera is available in starter kit and camera-only packages.

**Starter Kit**
https://www.framos.com/en/framos-depth-camera-d435e-starter-kit-22805

**Camera Only**
https://www.framos.com/en/framos-depth-camera-d435e-camera-only-22806

MartyG-RealSense, 2019-09-04:
Update: FRAMOS have released a promotional YouTube video for the D435e.

https://youtube.com/watch?v=_f890KiNVmA

closed 2019-10-04 by MartyG-RealSense

ShettyHarapanahalli, 2020-02-17:
IP 66 case for real sense, if anyone interested - 
https://www.autovimation.com/en/downloads/data-sheets-english/369-chameleon-xs-realsense-1/file

MartyG-RealSense, 2020-02-17:
@ShettyHarapanahalli   Thanks for the suggestion!   I have covered that product in a past announcement on this forum but there is no harm in highlighting it again   :)