realsenseai/librealsense #4770: Problems building unity

Soa11, 2019-08-31:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC  |
| SDK Version                            |  2.0                          | 
| Language                            |  Unity                      | 
| Segment			|  Installation                           | 

### Issue Description

Hi,
I'm trying to set up my D435 to work with unity. I'm following the steps on this page:
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/unity

I'm stuck on step 1 building the wrapper with cmake. I've got it installed but it is throwing this error:
`CMake Error: The source directory "C:/Users/Soa/Documents/RealSense SDK and Unity Package/librealsense-master/librealsense-master/build/.5" does not exist.`

Any suggestions? I've looked everywhere I can for more information.

Thanks,
Soa

MartyG-RealSense, 2019-08-31:
If you are using Windows 10 then there is an installation short-cut so you do not have to build the wrapper yourself.  You can use the pre-made DLL library files that are supplied with the RealSense SDK.

https://forums.intel.com/s/question/0D70P0000068Yo0SAE

RealSenseCustomerSupport, 2019-09-25:
----------------------------------------------

Hi @Soa11,

Do you fix this build issue? The latest prebuilt libraries can be also found in the release page:
https://github.com/IntelRealSense/librealsense/releases/download/v2.28.1/Intel.RealSense.unitypackage
https://github.com/IntelRealSense/librealsense/releases/download/v2.28.1/Intel.RealSense.SDK-WIN10-2.28.1.1106.exe

closed 2019-10-03 by RealSenseCustomerSupport