realsenseai/librealsense #4771: Color and depth maps are not aligned when moving fast

badforlabor, 2019-09-01:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | 05.11.06.200 | 
| Operating System & Version |  Win10 | 
| Kernel Version (Linux Only)    |                                       | 
| Platform                                 | PC  |
| SDK Version                            | 2.250                         | 
| Language                            |  c++                         | 
| Segment			| VR                       | 

### Issue Description
Color and depth maps are not aligned when moving fast，As shown below.

### Is there any way to solve it?

[image]

MartyG-RealSense, 2019-09-01:
Another RealSense user encountered this issue in February 2019, though a solution was not reached in that case.

https://github.com/IntelRealSense/librealsense/issues/3221

Are you using one of the SDK sample programs such as **rs-align** or **align-depth2color.py** or are you using your own script, please?   If it is your own script, please post the code in a comment below.  Thanks!

dorodnic, 2019-09-01:
If misalignment is due to movement, first thing to check is whether the RGB and depth cameras are in hardware-sync. By default, RGB camera operate in FPS priority mode, meaning it can reduce FPS (to say 15) if there is not enough light. This will, in turn disable hardware sync. You can set Exposure Priority Mode in the Viewer or via code and configure the cameras to same FPS. This should reduce the effect significantly.

badforlabor, 2019-09-01:
> If misalignment is due to movement, first thing to check is whether the RGB and depth cameras are in hardware-sync. By default, RGB camera operate in FPS priority mode, meaning it can reduce FPS (to [...]

Awesome, that's right，It improves the situation by 90%，still has problem like this
[image]

### have any other solution?

badforlabor, 2019-09-01:
> Another RealSense user encountered this issue in February 2019, though a solution was not reached in that case.
> 
> #3221
> 
> Are you using one of the SDK sample programs such as **rs-align** or **align-depth2color.py** or are you using your own script, please? In the case linked to above, the user was using **rs-align**.
> 
> If it is your own script, please post the code in a comment below. Thanks!

yes, I'm using sample align.vcxproj in SDK.

MartyG-RealSense, 2019-09-01:
Apologies for the delay in responding further.   I have been carefully considering and researching your case.

A test for whether the source of your problem is with the **rs-align** program may be to do a color to depth alignment in the RealSense Viewer's 3D point cloud mode and see if the same separation between color and depth occurs when you do fast motion.  The link below contains information about how to do this in the RealSense Viewer.

https://forums.intel.com/s/question/0D70P000006AebbSAC/pixels-mismatch

badforlabor, 2019-09-02:
> Apologies for the delay in responding further. I have been carefully considering and researching your case.
>
> A test for whether the source of your problem is with the **rs-align** program may be to [...]

I will give it a try, thank you very much.

closed 2019-09-02 by badforlabor

GrainsofSalt, 2020-02-14:
When in place, the depth and rgb are aligned perfectly. As the camera starts to move faster (in a straight line), the two images start to dis-align proportional to the vertical position (meaning at the bottom they are still aligned, but at the top they are completely dis-aligned). The exposure on the rgb was set to 25 microseconds so that the rgb images taken wouldn't blur when going around 5mph. I have been trying to understand how the camera initialization variables intermingle, but the documentation seems to be quite bare. For instance, I was trying to figure out if when auto exposure is disabled, does that mean exposure priority mode becomes useless? In any case, I enabled exposure priority and this did not help. Any suggestions as to how I can align the rgb and depth images when the rgb camera has manually set low exposure time? Thanks.