realsenseai/librealsense #4775: I'm having trouble finding where the variable "_pixel_top_left_int" is assigned in src/proc/sse/sse-align.cpp

kgh475926, 2019-09-01:
I'm having trouble finding where the variable "_pixel_top_left_int" is  assigned in src/proc/sse/sse-align.cpp

Right now I understand that the vector _pixel_top_left_int is initialized in the constructor called image_transform::image_transform with the size of from.width * from.height. 

So now I am trying to figure out where the code modifies the values in the vector _pixel_top_left_int but I am lost. Could I get some help?

aangerma, 2019-09-19:
Hello,
in sse-align.cpp image_transform::align_other_to_depth_sse() line 329  and image_transform::align_depth_to_other_sse line 301  
we are calling to get_texture_map_sse() and move (byte*)_pixel_top_left_int.data() as parameter
inside get_texture_map_sse() we calculate the texture map from depth to other stream and set the result on _pixel_top_left_int.

ev-mp, 2019-09-26:
@kgh475926 , was the info by @aangerma useful ? 
Do you have any follow up questions?

closed 2019-10-12 by RealSenseCustomerSupport